mik3y/usb-serial-for-android #373: Baudrate is set on different index since FTDI driver rework

MichaelBorde, 2021-06-30:
I use this library in version `2.3.0` and it is very stable.

My device is FTDI FT2232C.

I noticed something different about baudrate since the rework about FTDI driver (https://github.com/mik3y/usb-serial-for-android/commit/e496195bd97e9e88565b77fa75eabe212f33b857#diff-84649144c208a77ffac3cc29a1b30057e84d946bad38dcacaea46ede5016b615).

In version `2.3.0`, baudrate is set using index 2: https://github.com/mik3y/usb-serial-for-android/blob/7423fd9d79b330544743477b235e7d424f7ceb97/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java#L326.

Since rework (to `3.4.0`), baudrate is set using index 0 and it does not work on my device: https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java#L223

When I force index to 2 in the source code, it works as before.

Thank you for your work anyway.

kai-morich, 2021-06-30:
you're right, FT2232C support got lost. please try `ft2232c` branch which worked with my FT2232H device

kai-morich, 2021-07-01:
please try new `ft2232c` branch, now with single line fix

MichaelBorde, 2021-07-01:
I have just tested your branch and it now works for my device :)

Thank you for your work and your reactivity.

Could give me some information about setting this baud rate. I don't understand well why sometimes the index is different or what is the meaning of "baurate with port".

kai-morich, 2021-07-01:
merged to master

kai-morich, 2021-07-02:
available with v3.4.1

closed 2021-07-02 by kai-morich