mik3y/usb-serial-for-android #379: write long data Fail

boraylog, 2021-07-29:
try {
            val data: ByteArray? = HexUtils.toByteArray(address)
            usbSerialPort.write(data, 5000)
        } catch (e: Exception) {
            ToastUtils.showShort("fail message!-->${e.message}")
        }
fail message!--> length=1000;index=1000
thanks

kai-morich, 2021-07-29:
Very likely your write timeout is to short for the amount of data. 
See https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#How_to_set_timeout_in_codereadcodecodewritecode_methods

boraylog, 2021-08-03:
The write timeout  than 15s,Still report this error.**length=1000;index=1000**  
Please tell me how to solve it. thanks

Write protocol information.
66 10 90 78 56 34 12 33 78 58 02 03 0F 30 00 00 00 7A 00 00 03 81 32 07 E0 0F 50 1A 00 3A 30 C0 B3 A4 F3 B7 00 40 EB 92 F2 B7 5F D3 BF 68 F0 B7 60 D3 BF 69 F0 B7 61 D3 BF 6A F0 B7 2E D3 BF 6B F0 B7 2D D3 BF 6C F0 B7 2C D3 BF 6D F0 B7 00 C2 B3 6E F0 B7 23 29 21 02 0C 20 81 FA 20 90 0A 20 3B 36 07 31 00 20 40 C1 B3 56 32 07 00 40 EB 18 00 37 A4 D3 BF E3 4F 50 0A 00 3A 92 D2 BF BA 36 07 E1 4F 50 06 00 32 40 C1 B3 5A 32 07 E0 0F 50 02 00 32 00 40 EB 0B 00 37 F0 98 80 80 02 78 D4 F0 20 13 CA B3 32 C4 B3 DE 40 90 70 C6 B3 CB 00 07 10 C0 B3 A4 F3 B7 10 C0 B3 00 80 FA 00 00 06 00 00 FA 11 C0 B3 00 32 21 67 2C 07 21 D3 BF 74 F2 B7 22 D3 BF 75 F2 B7 23 D3 BF 76 F2 B7 24 D3 BF 77 F2 B7 C0 C2 B3 78 F2 B7 00 40 EB 79 F2 B7 00 40 EB 7A F2 B7 00 40 EB 7B F2 B7 00 40 EB 7C F2 B7 C0 C2 B3 7D F2 B7 00 40 EB 7E F2 B7 00 40 EB 7F F2 B7 00 40 EB 80 F2 B7 00 40 EB 81 F2 B7 00 40 EB 82 F2 B7 00 40 EB 83 F2 B7 00 40 EB 84 F2 B7 00 40 EB 6B 05 07 E0 4F 50 03 00 32 84 D2 BF 00 04 A0 84 F2 B7 10 C0 B3 64 05 07 E1 4F 50 03 00 32 84 D2 BF 00 14 A0 84 F2 B7 60 C0 B3 D3 2C 07 E0 4F 50 07 00 32 60 C0 B3 02 2D 07 41 28 21 91 40 78 00 C4 B3 00 C0 70 84 F2 B7 40 C0 B3 C8 2C 07 E0 4F 50 05 00 32 40 C0 B3 F7 2C 07 40 C0 B3 85 F2 B7 02 00 37 00 40 EB 85 F2 B7 F7 42 07 E0 4F 50 03 00 32 40 C0 B3 86 F2 B7 02 00 37 00 40 EB 86 F2 B7 00 80 FA 00 00 06 06 00 FA 20 47 98 31 47 98 42 47 98 53 47 98 00 40 EB 00 4F 78 E0 CF B3 80 F0 B7 E0 CF B3 81 F0 B7 80 C6 B3 82 F0 B7 00 C1 B3 83 F0 B7 00 40 EB 00 4F 78 0C 00 37 9E 40 78 01 80 FB 64 01 40 1E 40 78 80 80 FB 00 2A 21 00 80 40 10 40 78 01 08 21 81 01 41 80 48 78 1E 4F E8 1E 40 78 E6 4F 50 F1 FF 36 2E 40 90 8B F0 B7 3E 40 90 8C F0 B7 40 C0 B3 8D F0 B7 4E 40 90 8E F0 B7 5E 40 90 8F F0 B7 9B D2 BF 90 F0 B7 9C D2 BF 91 F0 B7 3E 40 90 00 80 FB 61 00 60 00 40 78 E0 4F 50 0C 00 32 00 40 EB 6B 02 07 8D D0 BF 00 80 FB 44 00 DE 00 40 78 00 80 FB 00 00 E8 00 40 78 41 00 20 01 00 DD 8D F0 B7 8D D0 BF 00 80 FB EE 00 40 00 08 21 80 80 40 00 40 EB 80 48 78 20 C0 B3 00 4F 78 13 00 37 8D D0 BF 00 80 FB 6E 01 40 8D D0 BF 00 80 FB EE 00 40 00 08 21 00 80 40 90 40 78 1E 40 78 80 81 FB 00 08 21 00 80 41 10 40 78 00 C0 40 01 08 21 81 00 41 80 48 78 1E 4F E8 1E 40 78 80 80 FB 8D D0 BF 00 80 FB 6E 00 40 80 8F 50 E6 FF 35 8D D0 BF 00 80 FB EF 01 40 00 08 21 80 80 40 60 C1 B3 80 48 78 8D D0 BF 00 80 FB 70 00 40 80 94 88 00 80 FA 00 10 06 0C 00 FA 4E 31 30 31 55 33 A1 6C 16

boraylog, 2021-08-03:
> Very likely your write timeout is to short for the amount of data.
> See https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#How_to_set_timeout_in_codereadcodecodewritecode_methods

> Very likely your write timeout is to short for the amount of data.
> See https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#How_to_set_timeout_in_codereadcodecodewritecode_methods

The write timeout than 15s,Still report this error.length=1000;index=1000
Please tell me how to solve it. thanks

Write protocol information.
66 10 90 78 56 34 12 33 78 58 02 03 0F 30 00 00 00 7A 00 00 03 81 32 07 E0 0F 50 1A 00 3A 30 C0 B3 A4 F3 B7 00 40 EB 92 F2 B7 5F D3 BF 68 F0 B7 60 D3 BF 69 F0 B7 61 D3 BF 6A F0 B7 2E D3 BF 6B F0 B7 2D D3 BF 6C F0 B7 2C D3 BF 6D F0 B7 00 C2 B3 6E F0 B7 23 29 21 02 0C 20 81 FA 20 90 0A 20 3B 36 07 31 00 20 40 C1 B3 56 32 07 00 40 EB 18 00 37 A4 D3 BF E3 4F 50 0A 00 3A 92 D2 BF BA 36 07 E1 4F 50 06 00 32 40 C1 B3 5A 32 07 E0 0F 50 02 00 32 00 40 EB 0B 00 37 F0 98 80 80 02 78 D4 F0 20 13 CA B3 32 C4 B3 DE 40 90 70 C6 B3 CB 00 07 10 C0 B3 A4 F3 B7 10 C0 B3 00 80 FA 00 00 06 00 00 FA 11 C0 B3 00 32 21 67 2C 07 21 D3 BF 74 F2 B7 22 D3 BF 75 F2 B7 23 D3 BF 76 F2 B7 24 D3 BF 77 F2 B7 C0 C2 B3 78 F2 B7 00 40 EB 79 F2 B7 00 40 EB 7A F2 B7 00 40 EB 7B F2 B7 00 40 EB 7C F2 B7 C0 C2 B3 7D F2 B7 00 40 EB 7E F2 B7 00 40 EB 7F F2 B7 00 40 EB 80 F2 B7 00 40 EB 81 F2 B7 00 40 EB 82 F2 B7 00 40 EB 83 F2 B7 00 40 EB 84 F2 B7 00 40 EB 6B 05 07 E0 4F 50 03 00 32 84 D2 BF 00 04 A0 84 F2 B7 10 C0 B3 64 05 07 E1 4F 50 03 00 32 84 D2 BF 00 14 A0 84 F2 B7 60 C0 B3 D3 2C 07 E0 4F 50 07 00 32 60 C0 B3 02 2D 07 41 28 21 91 40 78 00 C4 B3 00 C0 70 84 F2 B7 40 C0 B3 C8 2C 07 E0 4F 50 05 00 32 40 C0 B3 F7 2C 07 40 C0 B3 85 F2 B7 02 00 37 00 40 EB 85 F2 B7 F7 42 07 E0 4F 50 03 00 32 40 C0 B3 86 F2 B7 02 00 37 00 40 EB 86 F2 B7 00 80 FA 00 00 06 06 00 FA 20 47 98 31 47 98 42 47 98 53 47 98 00 40 EB 00 4F 78 E0 CF B3 80 F0 B7 E0 CF B3 81 F0 B7 80 C6 B3 82 F0 B7 00 C1 B3 83 F0 B7 00 40 EB 00 4F 78 0C 00 37 9E 40 78 01 80 FB 64 01 40 1E 40 78 80 80 FB 00 2A 21 00 80 40 10 40 78 01 08 21 81 01 41 80 48 78 1E 4F E8 1E 40 78 E6 4F 50 F1 FF 36 2E 40 90 8B F0 B7 3E 40 90 8C F0 B7 40 C0 B3 8D F0 B7 4E 40 90 8E F0 B7 5E 40 90 8F F0 B7 9B D2 BF 90 F0 B7 9C D2 BF 91 F0 B7 3E 40 90 00 80 FB 61 00 60 00 40 78 E0 4F 50 0C 00 32 00 40 EB 6B 02 07 8D D0 BF 00 80 FB 44 00 DE 00 40 78 00 80 FB 00 00 E8 00 40 78 41 00 20 01 00 DD 8D F0 B7 8D D0 BF 00 80 FB EE 00 40 00 08 21 80 80 40 00 40 EB 80 48 78 20 C0 B3 00 4F 78 13 00 37 8D D0 BF 00 80 FB 6E 01 40 8D D0 BF 00 80 FB EE 00 40 00 08 21 00 80 40 90 40 78 1E 40 78 80 81 FB 00 08 21 00 80 41 10 40 78 00 C0 40 01 08 21 81 00 41 80 48 78 1E 4F E8 1E 40 78 80 80 FB 8D D0 BF 00 80 FB 6E 00 40 80 8F 50 E6 FF 35 8D D0 BF 00 80 FB EF 01 40 00 08 21 80 80 40 60 C1 B3 80 48 78 8D D0 BF 00 80 FB 70 00 40 80 94 88 00 80 FA 00 10 06 0C 00 FA 4E 31 30 31 55 33 A1 6C 16

boraylog, 2021-08-03:
> Very likely your write timeout is to short for the amount of data.
> See https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#How_to_set_timeout_in_codereadcodecodewritecode_methods

The write timeout than 15s,Still report this error.length=1000;index=1000
Please tell me how to solve it. thanks

Write protocol information.
66 10 90 78 56 34 12 33 78 58 02 03 0F 30 00 00 00 7A 00 00 03 81 32 07 E0 0F 50 1A 00 3A 30 C0 B3 A4 F3 B7 00 40 EB 92 F2 B7 5F D3 BF 68 F0 B7 60 D3 BF 69 F0 B7 61 D3 BF 6A F0 B7 2E D3 BF 6B F0 B7 2D D3 BF 6C F0 B7 2C D3 BF 6D F0 B7 00 C2 B3 6E F0 B7 23 29 21 02 0C 20 81 FA 20 90 0A 20 3B 36 07 31 00 20 40 C1 B3 56 32 07 00 40 EB 18 00 37 A4 D3 BF E3 4F 50 0A 00 3A 92 D2 BF BA 36 07 E1 4F 50 06 00 32 40 C1 B3 5A 32 07 E0 0F 50 02 00 32 00 40 EB 0B 00 37 F0 98 80 80 02 78 D4 F0 20 13 CA B3 32 C4 B3 DE 40 90 70 C6 B3 CB 00 07 10 C0 B3 A4 F3 B7 10 C0 B3 00 80 FA 00 00 06 00 00 FA 11 C0 B3 00 32 21 67 2C 07 21 D3 BF 74 F2 B7 22 D3 BF 75 F2 B7 23 D3 BF 76 F2 B7 24 D3 BF 77 F2 B7 C0 C2 B3 78 F2 B7 00 40 EB 79 F2 B7 00 40 EB 7A F2 B7 00 40 EB 7B F2 B7 00 40 EB 7C F2 B7 C0 C2 B3 7D F2 B7 00 40 EB 7E F2 B7 00 40 EB 7F F2 B7 00 40 EB 80 F2 B7 00 40 EB 81 F2 B7 00 40 EB 82 F2 B7 00 40 EB 83 F2 B7 00 40 EB 84 F2 B7 00 40 EB 6B 05 07 E0 4F 50 03 00 32 84 D2 BF 00 04 A0 84 F2 B7 10 C0 B3 64 05 07 E1 4F 50 03 00 32 84 D2 BF 00 14 A0 84 F2 B7 60 C0 B3 D3 2C 07 E0 4F 50 07 00 32 60 C0 B3 02 2D 07 41 28 21 91 40 78 00 C4 B3 00 C0 70 84 F2 B7 40 C0 B3 C8 2C 07 E0 4F 50 05 00 32 40 C0 B3 F7 2C 07 40 C0 B3 85 F2 B7 02 00 37 00 40 EB 85 F2 B7 F7 42 07 E0 4F 50 03 00 32 40 C0 B3 86 F2 B7 02 00 37 00 40 EB 86 F2 B7 00 80 FA 00 00 06 06 00 FA 20 47 98 31 47 98 42 47 98 53 47 98 00 40 EB 00 4F 78 E0 CF B3 80 F0 B7 E0 CF B3 81 F0 B7 80 C6 B3 82 F0 B7 00 C1 B3 83 F0 B7 00 40 EB 00 4F 78 0C 00 37 9E 40 78 01 80 FB 64 01 40 1E 40 78 80 80 FB 00 2A 21 00 80 40 10 40 78 01 08 21 81 01 41 80 48 78 1E 4F E8 1E 40 78 E6 4F 50 F1 FF 36 2E 40 90 8B F0 B7 3E 40 90 8C F0 B7 40 C0 B3 8D F0 B7 4E 40 90 8E F0 B7 5E 40 90 8F F0 B7 9B D2 BF 90 F0 B7 9C D2 BF 91 F0 B7 3E 40 90 00 80 FB 61 00 60 00 40 78 E0 4F 50 0C 00 32 00 40 EB 6B 02 07 8D D0 BF 00 80 FB 44 00 DE 00 40 78 00 80 FB 00 00 E8 00 40 78 41 00 20 01 00 DD 8D F0 B7 8D D0 BF 00 80 FB EE 00 40 00 08 21 80 80 40 00 40 EB 80 48 78 20 C0 B3 00 4F 78 13 00 37 8D D0 BF 00 80 FB 6E 01 40 8D D0 BF 00 80 FB EE 00 40 00 08 21 00 80 40 90 40 78 1E 40 78 80 81 FB 00 08 21 00 80 41 10 40 78 00 C0 40 01 08 21 81 00 41 80 48 78 1E 4F E8 1E 40 78 80 80 FB 8D D0 BF 00 80 FB 6E 00 40 80 8F 50 E6 FF 35 8D D0 BF 00 80 FB EF 01 40 00 08 21 80 80 40 60 C1 B3 80 48 78 8D D0 BF 00 80 FB 70 00 40 80 94 88 00 80 FA 00 10 06 0C 00 FA 4E 31 30 31 55 33 A1 6C 16

kai-morich, 2021-08-03:
what is your baud rate and data length?

boraylog, 2021-08-04:
> What is your baud rate, data length and device type?
> Please provide full exception including call stack.

baud rate:2400 data length: 2393  device type: OTG 
**write data:**
fun writeData(pro: String, usbSerialPort: UsbSerialPort) {
        try {
            val data: ByteArray? = HexUtils.toByteArray(pro)
            usbSerialPort.write(data, 20000)
        } catch (e: Exception) {
            ToastUtils.showShort("fail message-->${e.message}")
        }
    }
**receive data**
 override fun onNewData(data: ByteArray) {
        mainLooper.post {
            receive(data)
        }
    }

full exception Still report this error.**length=1000;index=1000**

--------------------------------------------

Run with your example：
            byte[] data = (str + '\n').getBytes();
            SpannableStringBuilder spn = new SpannableStringBuilder();
            spn.append("send " + data.length + " bytes\n");
            spn.append(HexDump.dumpHexString(data)).append("\n");
            spn.setSpan(new ForegroundColorSpan(getResources().getColor(R.color.colorSendText)), 0, spn.length(), 
           Spannable.SPAN_EXCLUSIVE_EXCLUSIVE);
            receiveText.append(spn);
            usbSerialPort.write(data, WRITE_WAIT_MILLIS);

error writing 16384 byte at offset 32768 of total 275541, rc = -1

Write protocol information.
66 10 90 78 56 34 12 33 78 58 02 03 0F 30 00 00 00 7A 00 00 03 81 32 07 E0 0F 50 1A 00 3A 30 C0 B3 A4 F3 B7 00 40 EB 92 F2 B7 5F D3 BF 68 F0 B7 60 D3 BF 69 F0 B7 61 D3 BF 6A F0 B7 2E D3 BF 6B F0 B7 2D D3 BF 6C F0 B7 2C D3 BF 6D F0 B7 00 C2 B3 6E F0 B7 23 29 21 02 0C 20 81 FA 20 90 0A 20 3B 36 07 31 00 20 40 C1 B3 56 32 07 00 40 EB 18 00 37 A4 D3 BF E3 4F 50 0A 00 3A 92 D2 BF BA 36 07 E1 4F 50 06 00 32 40 C1 B3 5A 32 07 E0 0F 50 02 00 32 00 40 EB 0B 00 37 F0 98 80 80 02 78 D4 F0 20 13 CA B3 32 C4 B3 DE 40 90 70 C6 B3 CB 00 07 10 C0 B3 A4 F3 B7 10 C0 B3 00 80 FA 00 00 06 00 00 FA 11 C0 B3 00 32 21 67 2C 07 21 D3 BF 74 F2 B7 22 D3 BF 75 F2 B7 23 D3 BF 76 F2 B7 24 D3 BF 77 F2 B7 C0 C2 B3 78 F2 B7 00 40 EB 79 F2 B7 00 40 EB 7A F2 B7 00 40 EB 7B F2 B7 00 40 EB 7C F2 B7 C0 C2 B3 7D F2 B7 00 40 EB 7E F2 B7 00 40 EB 7F F2 B7 00 40 EB 80 F2 B7 00 40 EB 81 F2 B7 00 40 EB 82 F2 B7 00 40 EB 83 F2 B7 00 40 EB 84 F2 B7 00 40 EB 6B 05 07 E0 4F 50 03 00 32 84 D2 BF 00 04 A0 84 F2 B7 10 C0 B3 64 05 07 E1 4F 50 03 00 32 84 D2 BF 00 14 A0 84 F2 B7 60 C0 B3 D3 2C 07 E0 4F 50 07 00 32 60 C0 B3 02 2D 07 41 28 21 91 40 78 00 C4 B3 00 C0 70 84 F2 B7 40 C0 B3 C8 2C 07 E0 4F 50 05 00 32 40 C0 B3 F7 2C 07 40 C0 B3 85 F2 B7 02 00 37 00 40 EB 85 F2 B7 F7 42 07 E0 4F 50 03 00 32 40 C0 B3 86 F2 B7 02 00 37 00 40 EB 86 F2 B7 00 80 FA 00 00 06 06 00 FA 20 47 98 31 47 98 42 47 98 53 47 98 00 40 EB 00 4F 78 E0 CF B3 80 F0 B7 E0 CF B3 81 F0 B7 80 C6 B3 82 F0 B7 00 C1 B3 83 F0 B7 00 40 EB 00 4F 78 0C 00 37 9E 40 78 01 80 FB 64 01 40 1E 40 78 80 80 FB 00 2A 21 00 80 40 10 40 78 01 08 21 81 01 41 80 48 78 1E 4F E8 1E 40 78 E6 4F 50 F1 FF 36 2E 40 90 8B F0 B7 3E 40 90 8C F0 B7 40 C0 B3 8D F0 B7 4E 40 90 8E F0 B7 5E 40 90 8F F0 B7 9B D2 BF 90 F0 B7 9C D2 BF 91 F0 B7 3E 40 90 00 80 FB 61 00 60 00 40 78 E0 4F 50 0C 00 32 00 40 EB 6B 02 07 8D D0 BF 00 80 FB 44 00 DE 00 40 78 00 80 FB 00 00 E8 00 40 78 41 00 20 01 00 DD 8D F0 B7 8D D0 BF 00 80 FB EE 00 40 00 08 21 80 80 40 00 40 EB 80 48 78 20 C0 B3 00 4F 78 13 00 37 8D D0 BF 00 80 FB 6E 01 40 8D D0 BF 00 80 FB EE 00 40 00 08 21 00 80 40 90 40 78 1E 40 78 80 81 FB 00 08 21 00 80 41 10 40 78 00 C0 40 01 08 21 81 00 41 80 48 78 1E 4F E8 1E 40 78 80 80 FB 8D D0 BF 00 80 FB 6E 00 40 80 8F 50 E6 FF 35 8D D0 BF 00 80 FB EF 01 40 00 08 21 80 80 40 60 C1 B3 80 48 78 8D D0 BF 00 80 FB 70 00 40 80 94 88 00 80 FA 00 10 06 0C 00 FA 4E 31 30 31 55 33 A1 6C 16

boraylog, 2021-08-04:
> What is your baud rate, data length and device type?
> Please provide full exception including call stack.

baud rate:2400 data length: 2393  device type: OTG 
**write data:**
fun writeData(pro: String, usbSerialPort: UsbSerialPort) {
        try {
            val data: ByteArray? = HexUtils.toByteArray(pro)
            usbSerialPort.write(data, 20000)
        } catch (e: Exception) {
            ToastUtils.showShort("fail message-->${e.message}")
        }
    }
**receive data**
 override fun onNewData(data: ByteArray) {
        mainLooper.post {
            receive(data)
        }
    }

full exception Still report this error.**length=1000;index=1000**

--------------------------------------------

Run with your example：
            byte[] data = (str + '\n').getBytes();
            SpannableStringBuilder spn = new SpannableStringBuilder();
            spn.append("send " + data.length + " bytes\n");
            spn.append(HexDump.dumpHexString(data)).append("\n");
            spn.setSpan(new ForegroundColorSpan(getResources().getColor(R.color.colorSendText)), 0, spn.length(), 
           Spannable.SPAN_EXCLUSIVE_EXCLUSIVE);
            receiveText.append(spn);
            usbSerialPort.write(data, WRITE_WAIT_MILLIS);

error writing 16384 byte at offset 32768 of total 275541, rc = -1

Write data information.
66 10 90 78 56 34 12 33 78 58 02 03 0F 30 00 00 00 7A 00 00 03 81 32 07 E0 0F 50 1A 00 3A 30 C0 B3 A4 F3 B7 00 40 EB 92 F2 B7 5F D3 BF 68 F0 B7 60 D3 BF 69 F0 B7 61 D3 BF 6A F0 B7 2E D3 BF 6B F0 B7 2D D3 BF 6C F0 B7 2C D3 BF 6D F0 B7 00 C2 B3 6E F0 B7 23 29 21 02 0C 20 81 FA 20 90 0A 20 3B 36 07 31 00 20 40 C1 B3 56 32 07 00 40 EB 18 00 37 A4 D3 BF E3 4F 50 0A 00 3A 92 D2 BF BA 36 07 E1 4F 50 06 00 32 40 C1 B3 5A 32 07 E0 0F 50 02 00 32 00 40 EB 0B 00 37 F0 98 80 80 02 78 D4 F0 20 13 CA B3 32 C4 B3 DE 40 90 70 C6 B3 CB 00 07 10 C0 B3 A4 F3 B7 10 C0 B3 00 80 FA 00 00 06 00 00 FA 11 C0 B3 00 32 21 67 2C 07 21 D3 BF 74 F2 B7 22 D3 BF 75 F2 B7 23 D3 BF 76 F2 B7 24 D3 BF 77 F2 B7 C0 C2 B3 78 F2 B7 00 40 EB 79 F2 B7 00 40 EB 7A F2 B7 00 40 EB 7B F2 B7 00 40 EB 7C F2 B7 C0 C2 B3 7D F2 B7 00 40 EB 7E F2 B7 00 40 EB 7F F2 B7 00 40 EB 80 F2 B7 00 40 EB 81 F2 B7 00 40 EB 82 F2 B7 00 40 EB 83 F2 B7 00 40 EB 84 F2 B7 00 40 EB 6B 05 07 E0 4F 50 03 00 32 84 D2 BF 00 04 A0 84 F2 B7 10 C0 B3 64 05 07 E1 4F 50 03 00 32 84 D2 BF 00 14 A0 84 F2 B7 60 C0 B3 D3 2C 07 E0 4F 50 07 00 32 60 C0 B3 02 2D 07 41 28 21 91 40 78 00 C4 B3 00 C0 70 84 F2 B7 40 C0 B3 C8 2C 07 E0 4F 50 05 00 32 40 C0 B3 F7 2C 07 40 C0 B3 85 F2 B7 02 00 37 00 40 EB 85 F2 B7 F7 42 07 E0 4F 50 03 00 32 40 C0 B3 86 F2 B7 02 00 37 00 40 EB 86 F2 B7 00 80 FA 00 00 06 06 00 FA 20 47 98 31 47 98 42 47 98 53 47 98 00 40 EB 00 4F 78 E0 CF B3 80 F0 B7 E0 CF B3 81 F0 B7 80 C6 B3 82 F0 B7 00 C1 B3 83 F0 B7 00 40 EB 00 4F 78 0C 00 37 9E 40 78 01 80 FB 64 01 40 1E 40 78 80 80 FB 00 2A 21 00 80 40 10 40 78 01 08 21 81 01 41 80 48 78 1E 4F E8 1E 40 78 E6 4F 50 F1 FF 36 2E 40 90 8B F0 B7 3E 40 90 8C F0 B7 40 C0 B3 8D F0 B7 4E 40 90 8E F0 B7 5E 40 90 8F F0 B7 9B D2 BF 90 F0 B7 9C D2 BF 91 F0 B7 3E 40 90 00 80 FB 61 00 60 00 40 78 E0 4F 50 0C 00 32 00 40 EB 6B 02 07 8D D0 BF 00 80 FB 44 00 DE 00 40 78 00 80 FB 00 00 E8 00 40 78 41 00 20 01 00 DD 8D F0 B7 8D D0 BF 00 80 FB EE 00 40 00 08 21 80 80 40 00 40 EB 80 48 78 20 C0 B3 00 4F 78 13 00 37 8D D0 BF 00 80 FB 6E 01 40 8D D0 BF 00 80 FB EE 00 40 00 08 21 00 80 40 90 40 78 1E 40 78 80 81 FB 00 08 21 00 80 41 10 40 78 00 C0 40 01 08 21 81 00 41 80 48 78 1E 4F E8 1E 40 78 80 80 FB 8D D0 BF 00 80 FB 6E 00 40 80 8F 50 E6 FF 35 8D D0 BF 00 80 FB EF 01 40 00 08 21 80 80 40 60 C1 B3 80 48 78 8D D0 BF 00 80 FB 70 00 40 80 94 88 00 80 FA 00 10 06 0C 00 FA 4E 31 30 31 55 33 A1 6C 16

thanks

boraylog, 2021-08-04:
buffer = null;
        synchronized (mWriteBufferLock) {
            len = mWriteBuffer.position();  **// is 0**
            if (len > 0) {
                buffer = new byte[len];
                mWriteBuffer.rewind();
                mWriteBuffer.get(buffer, 0, len);
                mWriteBuffer.clear();
            }
        }
        if (buffer != null) {
            if (DEBUG) {
                Log.d(TAG, "Writing data len=" + len);
            }
            mSerialPort.write(buffer, mWriteTimeout);
        }

boraylog, 2021-08-04:
buffer = null;
synchronized (mWriteBufferLock) {
len = mWriteBuffer.position(); // is 0
if (len > 0) {
buffer = new byte[len];
mWriteBuffer.rewind();
mWriteBuffer.get(buffer, 0, len);
mWriteBuffer.clear();
}
}
if (buffer != null) {
if (DEBUG) {
Log.d(TAG, "Writing data len=" + len);
}
mSerialPort.write(buffer, mWriteTimeout);
}

len = mWriteBuffer.position();  // is 0

boraylog, 2021-08-04:
What is your baud rate, data length and device type?
Please provide full exception including call stack.

baud rate:2400 data length: 2393 device type: OTG
write data:
fun writeData(pro: String, usbSerialPort: UsbSerialPort) {
try {
val data: ByteArray? = HexUtils.toByteArray(pro)
usbSerialPort.write(data, 20000)
} catch (e: Exception) {
ToastUtils.showShort("fail message-->${e.message}")
}
}
receive data
override fun onNewData(data: ByteArray) {
mainLooper.post {
receive(data)
}
}

full exception Still report this error.length=1000;index=1000

--------------------------------------------------------

Run with your example：
byte[] data = (str + '\n').getBytes();
SpannableStringBuilder spn = new SpannableStringBuilder();
spn.append("send " + data.length + " bytes\n");
spn.append(HexDump.dumpHexString(data)).append("\n");
spn.setSpan(new ForegroundColorSpan(getResources().getColor(R.color.colorSendText)), 0, spn.length(),
Spannable.SPAN_EXCLUSIVE_EXCLUSIVE);
receiveText.append(spn);
usbSerialPort.write(data, WRITE_WAIT_MILLIS);
**Error writing 2396 bytes offset 0 of total 2396**

Write data information.
66 10 90 78 56 34 12 33 78 58 02 03 0F 30 00 00 00 7A 00 00 03 81 32 07 E0 0F 50 1A 00 3A 30 C0 B3 A4 F3 B7 00 40 EB 92 F2 B7 5F D3 BF 68 F0 B7 60 D3 BF 69 F0 B7 61 D3 BF 6A F0 B7 2E D3 BF 6B F0 B7 2D D3 BF 6C F0 B7 2C D3 BF 6D F0 B7 00 C2 B3 6E F0 B7 23 29 21 02 0C 20 81 FA 20 90 0A 20 3B 36 07 31 00 20 40 C1 B3 56 32 07 00 40 EB 18 00 37 A4 D3 BF E3 4F 50 0A 00 3A 92 D2 BF BA 36 07 E1 4F 50 06 00 32 40 C1 B3 5A 32 07 E0 0F 50 02 00 32 00 40 EB 0B 00 37 F0 98 80 80 02 78 D4 F0 20 13 CA B3 32 C4 B3 DE 40 90 70 C6 B3 CB 00 07 10 C0 B3 A4 F3 B7 10 C0 B3 00 80 FA 00 00 06 00 00 FA 11 C0 B3 00 32 21 67 2C 07 21 D3 BF 74 F2 B7 22 D3 BF 75 F2 B7 23 D3 BF 76 F2 B7 24 D3 BF 77 F2 B7 C0 C2 B3 78 F2 B7 00 40 EB 79 F2 B7 00 40 EB 7A F2 B7 00 40 EB 7B F2 B7 00 40 EB 7C F2 B7 C0 C2 B3 7D F2 B7 00 40 EB 7E F2 B7 00 40 EB 7F F2 B7 00 40 EB 80 F2 B7 00 40 EB 81 F2 B7 00 40 EB 82 F2 B7 00 40 EB 83 F2 B7 00 40 EB 84 F2 B7 00 40 EB 6B 05 07 E0 4F 50 03 00 32 84 D2 BF 00 04 A0 84 F2 B7 10 C0 B3 64 05 07 E1 4F 50 03 00 32 84 D2 BF 00 14 A0 84 F2 B7 60 C0 B3 D3 2C 07 E0 4F 50 07 00 32 60 C0 B3 02 2D 07 41 28 21 91 40 78 00 C4 B3 00 C0 70 84 F2 B7 40 C0 B3 C8 2C 07 E0 4F 50 05 00 32 40 C0 B3 F7 2C 07 40 C0 B3 85 F2 B7 02 00 37 00 40 EB 85 F2 B7 F7 42 07 E0 4F 50 03 00 32 40 C0 B3 86 F2 B7 02 00 37 00 40 EB 86 F2 B7 00 80 FA 00 00 06 06 00 FA 20 47 98 31 47 98 42 47 98 53 47 98 00 40 EB 00 4F 78 E0 CF B3 80 F0 B7 E0 CF B3 81 F0 B7 80 C6 B3 82 F0 B7 00 C1 B3 83 F0 B7 00 40 EB 00 4F 78 0C 00 37 9E 40 78 01 80 FB 64 01 40 1E 40 78 80 80 FB 00 2A 21 00 80 40 10 40 78 01 08 21 81 01 41 80 48 78 1E 4F E8 1E 40 78 E6 4F 50 F1 FF 36 2E 40 90 8B F0 B7 3E 40 90 8C F0 B7 40 C0 B3 8D F0 B7 4E 40 90 8E F0 B7 5E 40 90 8F F0 B7 9B D2 BF 90 F0 B7 9C D2 BF 91 F0 B7 3E 40 90 00 80 FB 61 00 60 00 40 78 E0 4F 50 0C 00 32 00 40 EB 6B 02 07 8D D0 BF 00 80 FB 44 00 DE 00 40 78 00 80 FB 00 00 E8 00 40 78 41 00 20 01 00 DD 8D F0 B7 8D D0 BF 00 80 FB EE 00 40 00 08 21 80 80 40 00 40 EB 80 48 78 20 C0 B3 00 4F 78 13 00 37 8D D0 BF 00 80 FB 6E 01 40 8D D0 BF 00 80 FB EE 00 40 00 08 21 00 80 40 90 40 78 1E 40 78 80 81 FB 00 08 21 00 80 41 10 40 78 00 C0 40 01 08 21 81 00 41 80 48 78 1E 4F E8 1E 40 78 80 80 FB 8D D0 BF 00 80 FB 6E 00 40 80 8F 50 E6 FF 35 8D D0 BF 00 80 FB EF 01 40 00 08 21 80 80 40 60 C1 B3 80 48 78 8D D0 BF 00 80 FB 70 00 40 80 94 88 00 80 FA 00 10 06 0C 00 FA 4E 31 30 31 55 33 A1 6C 16

thanks

kai-morich, 2021-08-04:
You pasted a code snippet, but not the exception type + callstack. You can print to logcat with `e.printStackTrace();` or if you don't catch it at all, it will be shown after app crash in Android Studio logcat window.

What is your device type? Should be something like CH340, FT232, ...

boraylog, 2021-08-04:
thank you very much.
I found that the set array is out of bounds.
thanks again.

closed 2021-08-04 by boraylog