mik3y/usb-serial-for-android #380: Connection Issue with Raspberry Pi Controller

PemmasaniRajesh, 2021-08-10:
I've used this  #376  to connect android app with Raspberry Pi. My Pi USB port has 5V. Whenever I connect the sample app to the PI, only android.hardware.usb.action.USB_STATE is firing two times, USB_DEVICE_ATTACHED event hasn't fired atleast once.

Do you've any idea about this ?

kai-morich, 2021-08-10:
how did you connect the RasPi to your Android device? 
With standard USB cable in RasPi USB-A port and USB micro/C connector in your Android device?

PemmasaniRajesh, 2021-08-11:
Yeah, I'm using standard USB cable only. USB-A end to Raspberry Pi and USB-C end to Android device.

kai-morich, 2021-08-11:
With a standard cable, the RasPi is in host mode and the Android device is in slave mode. This does not support serial connection.

For a serial connection the Android device has to be in host mode, typically with an OTG adapter. But the RasPi USB A sockets typically cannot run in slave mode, except very old RasPis where the ports are directly connected to the CPU.

You should connect a USB2serial module to the serial pins on the RasPi GPIO header (or connect 2 USB2serial modules)

closed 2021-09-08 by kai-morich