mik3y/usb-serial-for-android #382: App falls down with ProbeTable

SunSayIsCool, 2021-08-16:
When I start my service for CDC device port listening it falls down with this error:
java.lang.NoClassDefFoundError: Failed resolution of: Lcom/hoho/android/usbserial/driver/ProbeTable;

Can you advice me how to fix it?

kai-morich, 2021-08-17:
which version of the library do you use? do you compile it yourself?
Since v2.1.0 a [proguard-rules.pro entry](https://github.com/mik3y/usb-serial-for-android/blob/cd83951bd14031ff87f601acf2f72df38330cbbd/usbSerialForAndroid/proguard-rules.pro#L1)  should prevent elimination of classes.

SunSayIsCool, 2021-08-17:
I'm using 3.4.2 version and add to my project by implementing dependency in Gradle

kai-morich, 2021-08-17:
please provide exception call stack

closed 2021-09-26 by kai-morich