mik3y/usb-serial-for-android #384: connection failed : ( RS232 - TypeC )

lh414035210, 2021-09-06:
Hi,   I  hava a problem, Hope the author can give the answer!  

       I  use RS232 - TypeC ,    connect  failed:   ControlTransfer 0x8 failed: -1
       
       I use   RS232-USB,  USB-TypeC   ， Success；

why  use    RS232 - TypeC failed?

[image]
[image]

kai-morich, 2021-09-08:
what kind of device did you attach?

lh414035210, 2021-09-08:
use my phone,     HUAWEI P40 PRO ,  android  10.0

kai-morich, 2021-09-08:
what kind of serial device did you attach to your phone?

lh414035210, 2021-09-08:
Nothing is installed， just using your demo

kai-morich, 2021-09-08:
without attached USB serial device (ft232, pl2303, ...) the app cannot work. 
probably you your phone as some internal USB device that is shown in the device list, but that is not a serial device.

lh414035210, 2021-09-08:
We can already communicate using RS232-USB + USB-Typec, I just want to know why RS232-TypeC can't

kai-morich, 2021-09-08:
still unclear what you want to connect.
Is your 'RS232-USB + USB-Typec' an RS232-to-USB-A cable + USB-OTG-Adapter?
Is your 'RS232-TypeC' is an RS232-to-USB-C cable? This will typically not work as an OTG-adapter is required to switch your android device into USB host mode.

kai-morich, 2021-10-12:
mentioned mandatory OTG adapter in [FAQ](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#my-usb-device-is-not-recognized)