mik3y/usb-serial-for-android #386: what is the maximum size of each data-tranfering?

SheStealsMyPenta, 2021-09-15:
Hi there , i am meeting a problem that the size of the data is about 86KB and we didn't separate the data, so i can't receive the data in a properly size. if it's invalid to put such large size data in one-time transfer ?

kai-morich, 2021-09-15:
thats ok. the data will be split into 16KB as old android versions do not support larger chunks
https://github.com/mik3y/usb-serial-for-android/blob/12095f6b942c654b241f5a4bcf596a0b82a58af9/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CommonUsbSerialPort.java#L29
and the USB layer then splits again into `getWriteEndpoint().getMaxPacketSize()` chunks which is 64 byte for many devices.

SheStealsMyPenta, 2021-09-15:
thanks for the guide ,  and i fixed the problem, that  was because i called Thread.sleep , so it would lose some raw data from the stream. lol

closed 2021-09-15 by SheStealsMyPenta