mik3y/usb-serial-for-android #389: Unnecessary output to logcat

janimm, 2021-09-22:
On each write to serial, logcat message is outputted

    D/CommonUsbSerialPort: Wrote #/# offset #/# timeout ###

This is on version 3.4.2.

Not a big problem but unnecessarily floods the logcat.

kai-morich, 2021-09-23:
with `adb shell setprop log.tag.CommonUsbSerialPort INFO` you can reduce the logging level since Android 8.0

janimm, 2021-09-24:
I'm aware of that. Problem to my opinion is that this unnecessarily floods the log on released software. I'd prefer a function that would toggle log output on/off.

kai-morich, 2021-09-24:
could be solved similar to this: 
https://github.com/mik3y/usb-serial-for-android/blob/d3198793861db418ee1ac33ef929d1ab49a8d878/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/util/SerialInputOutputManager.java#L25

janimm, 2021-09-24:
That would be perfect solution if you could add this to a future release.

closed 2021-10-02 by kai-morich