mik3y/usb-serial-for-android #395: CdcAcm driver does not work with baud rate 230400

janimm, 2021-11-01:
Device: NXP/Freescale.
VID: 0x1FC9
PID: 0x823D
Custom probe for CdcAcm driver.
Baud rate: 230400

Port connects but no data is received. The felHR85 serial library works with same setup.

kai-morich, 2021-11-01:
the CDC driver does not use any baud rate on its own. It sends all parameters set with `setParameters` to the USB device as SET_LINE_CODING request. Do these look different?

janimm, 2021-11-02:
They look very similar. Another library is setting the parameter bytes again for each individual setting, but otherwise it seems to be identical.

ezellanthony, 2022-01-27:
> Device: NXP/Freescale.
> VID: 0x1FC9
> PID: 0x823D
> Custom probe for CdcAcm driver.
> Baud rate: 230400
> 
> Port connects but no data is received. The felHR85 serial library works with same setup.

closed 2023-10-18 by kai-morich