mik3y/usb-serial-for-android #396: USB get_status request failed

omgljw, 2021-11-04:
Hi,
Please help me on this

when I used, I received the error USB get_status request failed, I made SerialInputOutputManager stop ,UsbSerialPort close.
after I connect again, but I received error on UsbSerialPort.open. I am getting below error
 java.io.IOException: Failed send cmd [init #1]
 <br> com.hoho.android.usbserial.driver.Ch34xSerialDriver$Ch340SerialPort.checkState(Ch34xSerialDriver.java:134)
 <br> com.hoho.android.usbserial.driver.Ch34xSerialDriver$Ch340SerialPort.initialize(Ch34xSerialDriver.java:168)
 <br> com.hoho.android.usbserial.driver.Ch34xSerialDriver$Ch340SerialPort.openInt(Ch34xSerialDriver.java:103)
 <br> com.hoho.android.usbserial.driver.CommonUsbSerialPort.open(CommonUsbSerialPort.java:109)

omgljw, 2021-11-04:
Please help me again.
At first, I can read data normally， but may be five hours later the above problems occur, It may always be normal

closed 2021-12-04 by kai-morich

omgljw, 2021-12-17:
this is my print log,peleas help me,Thanks
2021.12.17 13:50:18.239,发送指令成功
2021.12.17 13:50:18.310,0241422B30303032393034313703
2021.12.17 13:50:18.311,读取数据：2904
2021.12.17 13:50:18.739,发送指令成功
2021.12.17 13:50:18.790,0241422B30303032393034313703
2021.12.17 13:50:18.792,读取数据：2904
2021.12.17 13:50:19.240,发送指令成功
2021.12.17 13:50:19.271,0241422B30303032393034313703
2021.12.17 13:50:19.272,读取数据：2904
2021.12.17 13:50:19.740,发送指令成功
2021.12.17 13:50:19.830,0241422B30303032393034313703
2021.12.17 13:50:19.832,读取数据：2904
2021.12.17 13:50:20.241,发送指令成功
2021.12.17 13:50:20.311,0241422B30303032393034313703
2021.12.17 13:50:20.312,读取数据：2904
2021.12.17 13:50:20.731,读取错误：java.io.IOException: USB get_status request failed
2021.12.17 13:50:20.734,地磅连接断开
2021.12.17 13:50:22.130,地磅准备连接
2021.12.17 13:50:22.155,请检查地磅接口:java.io.IOException: ControlTransfer 0x8 failed: -1
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.outControlTransfer(ProlificSerialDriver.java:151)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.vendorOut(ProlificSerialDriver.java:162)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.purgeHwBuffers(ProlificSerialDriver.java:554)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.resetDevice(ProlificSerialDriver.java:166)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.openInt(ProlificSerialDriver.java:311)
 <br> com.hoho.android.usbserial.driver.CommonUsbSerialPort.open(CommonUsbSerialPort.java:109)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager.connect(UsbBalanceManager.kt:170)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager.access$connect(UsbBalanceManager.kt:54)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager$3.accept(UsbBalanceManager.kt:104)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager$3.accept(UsbBalanceManager.kt:54)
 <br> io.reactivex.internal.subscribers.LambdaSubscriber.onNext(LambdaSubscriber.java:65)
 <br> io.reactivex.internal.operators.flowable.FlowableInterval$IntervalSubscriber.run(FlowableInterval.java:93)
 <br> io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
 <br> java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:428)
 <br> java.util.concurrent.FutureTask.runAndReset(FutureTask.java:278)
 <br> java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:273)
 <br> java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
 <br> java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
 <br> java.lang.Thread.run(Thread.java:761)
2021.12.17 13:50:22.160,地磅连接断开
2021.12.17 13:50:23.161,异常次数1
2021.12.17 13:50:24.129,地磅准备连接
2021.12.17 13:50:24.147,请检查地磅接口:java.io.IOException: ControlTransfer 0x8 failed: -1
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.outControlTransfer(ProlificSerialDriver.java:151)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.vendorOut(ProlificSerialDriver.java:162)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.purgeHwBuffers(ProlificSerialDriver.java:554)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.resetDevice(ProlificSerialDriver.java:166)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.openInt(ProlificSerialDriver.java:311)
 <br> com.hoho.android.usbserial.driver.CommonUsbSerialPort.open(CommonUsbSerialPort.java:109)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager.connect(UsbBalanceManager.kt:170)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager.access$connect(UsbBalanceManager.kt:54)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager$3.accept(UsbBalanceManager.kt:104)
 <br> com.mtm.medicalwaste.dispose.balance.UsbBalanceManager$3.accept(UsbBalanceManager.kt:54)
 <br> io.reactivex.internal.subscribers.LambdaSubscriber.onNext(LambdaSubscriber.java:65)
 <br> io.reactivex.internal.operators.flowable.FlowableInterval$IntervalSubscriber.run(FlowableInterval.java:93)
 <br> io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
 <br> java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:428)
 <br> java.util.concurrent.FutureTask.runAndReset(FutureTask.java:278)
 <br> java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:273)
 <br> java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
 <br> java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
 <br> java.lang.Thread.run(Thread.java:761)
2021.12.17 13:50:24.152,地磅连接断开
2021.12.17 13:50:25.154,异常次数2
2021.12.17 13:50:26.128,地磅准备连接
2021.12.17 13:50:26.147,请检查地磅接口:java.io.IOException: ControlTransfer 0x8 failed: -1
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.outControlTransfer(ProlificSerialDriver.java:151)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.vendorOut(ProlificSerialDriver.java:162)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.purgeHwBuffers(ProlificSerialDriver.java:554)
 <br> com.hoho.android.usbserial.driver.ProlificSerialDriver$ProlificSerialPort.resetDevice(ProlificSerialDriver.java:166)
[... cut ...]
 <br> java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:428)
 <br> java.util.concurrent.FutureTask.runAndReset(FutureTask.java:278)
 <br> java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:273)
 <br> java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
 <br> java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
 <br> java.lang.Thread.run(Thread.java:761)
2021.12.17 13:50:26.152,地磅连接断开

Mohammedbinnazar, 2023-08-09:
@kai-morich Hi !
I also get "USB get_status request failed" after some time performing some transactions.

```
27360 24132 I androidpas: Read Length -> 1
08-02 18:36:02.644 27360 24132 I androidpas: Read Length -> 1
08-02 18:36:02.645 27360 24132 I androidpas: Read Length -> 120
08-02 18:36:02.666 27360 24132 I androidpas: [EPOS    NGPAS <- PED] ping
08-02 18:36:02.671 27360 27369 D UsbRequestJNI: close
08-02 18:36:04.108 27360 22876 I androidpas: [EPOS <- NGPAS    PED] error {"message":"Command timed out","status":101}
08-02 18:36:04.665 27360 24132 I androidpas: Read Length -> 0
08-02 18:36:05.092 27360 27614 I androidpas: [EPOS -> NGPAS    PED] getStatus()
08-02 18:36:05.096 27360 27614 I androidpas: [EPOS    NGPAS ---> PED] getStatus()
08-02 18:36:05.253 27360 24132 I androidpas: Read Length -> 413
08-02 18:36:05.254 27360 24132 I androidpas: [EPOS    NGPAS <- PED] transaction {"cardInserted":false,"complete":true,"dccAccepted":false,"dccOffered":false,"declined":false,"displayText":"NO TXN / SYSTEM IDLE","eposCheckCard":false,"inProgress":false,"offline":false,"signatureRequired":false,"success":false,"tokenTransaction":false}
08-02 18:36:05.255 27360 24132 I androidpas: [EPOS <- NGPAS    PED] transaction {"cardInserted":false,"complete":true,"dccAccepted":false,"dccOffered":false,"declined":false,"displayText":"NO TXN / SYSTEM IDLE","eposCheckCard":false,"inProgress":false,"offline":false,"signatureRequired":false,"success":false,"tokenTransaction":false}
08-02 18:36:06.310 27360 27614 I androidpas: [EPOS -> NGPAS    PED] getResult()
08-02 18:36:06.311 27360 27614 I androidpas: [EPOS    NGPAS ---> PED] getResult()
08-02 18:36:06.573 27360 24132 I androidpas: Read Length -> 0
08-02 18:36:06.627 27360 24132 I androidpas: Read Length -> 0
08-02 18:36:06.900 27360 24132 I androidpas: Read Length -> 0
08-02 18:36:06.904 27360 24132 I androidpas: Read Length -> 0
08-02 18:36:07.392 27360 24132 I androidpas: Read Length -> 0
08-02 18:36:07.655 27360 24132 I chatty  : uid=10154(com.trustpayments.ni.androidpas) Thread-71 identical 3 lines
08-02 18:36:07.667 27360 24132 I androidpas: Read Length -> 0
08-02 18:36:07.746 27360 24132 E androidpas: Exception caught while reading
08-02 18:36:07.746 27360 24132 E androidpas: **java.io.IOException: USB get_status request failed**
08-02 18:36:07.746 27360 24132 E androidpas: 	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.testConnection(Unknown Source:24)
08-02 18:36:07.746 27360 24132 E androidpas: 	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(Unknown Source:48)
08-02 18:36:07.746 27360 24132 E androidpas: 	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(Unknown Source:1)
08-02 18:36:07.746 27360 24132 E androidpas: 	at com.trustpayments.ni.androidpas.connection.ped.AndroidUsbSerialPedConnection$UsbCommsThread.readMessage(Unknown Source:43)
08-02 18:36:07.746 27360 24132 E androidpas: 	at com.trustpayments.ni.androidpas.comms.CommsThread.run(Unknown Source:7)
08-02 18:36:07.747 27360 24132 D UsbDeviceConnectionJNI: close
08-02 18:36:07.747 27360 24132 I androidpas: PED Connection Lost. Reconnecting
08-02 18:36:07.748 27360 24132 E androidpas: Interrupted Exception
08-02 18:36:07.748 27360 24132 E androidpas: java.lang.InterruptedException
08-02 18:36:07.748 27360 24132 E androidpas: 	at java.lang.Thread.sleep(Native Method)
08-02 18:36:07.748 27360 24132 E androidpas: 	at java.lang.Thread.sleep(Thread.java:373)
08-02 18:36:07.748 27360 24132 E androidpas: 	at java.lang.Thread.sleep(Thread.java:314)
08-02 18:36:07.748 27360 24132 E androidpas: 	at com.trustpayments.ni.androidpas.comms.CommsThread.run(Unknown Source:81)
08-02 18:36:07.798 27360 27418 I androidpas: Opening Connection!
08-02 18:36:07.801 27360 27418 I androidpas: USB Permission granted for device 0123456789ABCDEF
08-02 18:36:07.802 27360 27418 D CdcAcmSerialDriver: trying default interface logic
08-02 18:36:07.802 27360 27418 D CdcAcmSerialDriver: claiming interfaces, count=2
08-02 18:36:07.802 27360 27418 D CdcAcmSerialDriver: Control iface=UsbInterface[mId=0,mAlternateSetting=0,mName=CDC Abstract Control Model (ACM),mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[
08-02 18:36:07.802 27360 27418 D CdcAcmSerialDriver: UsbEndpoint[mAddress=130,mAttributes=3,mMaxPacketSize=10,mInterval=9]]
08-02 18:36:07.802 27360 27418 D CdcAcmSerialDriver: data iface=UsbInterface[mId=1,mAlternateSetting=0,mName=CDC ACM Data,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
08-02 18:36:07.802 27360 27418 D CdcAcmSerialDriver: UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0]
08-02 18:36:07.802 27360 27418 D CdcAcmSerialDriver: UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
08-02 18:36:07.802 27360 27418 D UsbRequestJNI: init
```
After this crash it opens the connection back but no data is being passed, no ping response also received from the device ?
This is my read part of the code

```
@Override
		protected Message readMessage() {
			byte[] read = new byte[65536];
			if (!port.isOpen()) {
				close();
				listener.commsClosed();
				return null;
			}

			while (!isInterrupted()) {
				try {
					int len = port.read(read, TIMEOUT);
					Log.i(TAG, "Read Length -> " + len);

					if (len > 0) {
[... 9 lines cut ...]
								bufStr.setLength(0);
								bufStr.append(bufParseStr.substring(bufParseStr.indexOf("}{") + 1));
							} else {
								bufStr.setLength(0);
							}
							return message;
						}
[... cut ...]
				} catch (JsonParseException | JsonMappingException ignored) {
					// NEX-10076 - NI do not want partial message alerts printing in the AndroidPAS logs, catch is kept here to keep functionality identical
					//Log.i(TAG,"Partial message received, waiting for remainder of message: "+ ignored.getMessage());
				} catch (Exception e) {
					Log.e(TAG, "Exception caught while reading", e);
					bufStr.setLength(0);
					close();
					listener.commsClosed();
					break;
				}
			}
			return null;
		}

```

joseph-lewis, 2024-02-23:
Was there ever a solution for this? It was closed but no answer was provided? I am receiving the same error after simply disconnecting my USB device, then on ATTACH it triggers my listener, which triggers code to connect to the usb device. It connects okay but immediately after crashes with the same crash stated above. This is now running v3.7.0? Any solutions would be appreciated, here is my error

```
Run ending due to exception: USB get_status request failed
java.io.IOException: USB get_status request failed at 
com.hoho.android.usbserial.driver.CommonUsbSerialPort.testConnection(CommonUsbSerialPort.java:167) at 
com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:213)at 
com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:172) at 
com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:225) at 
com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:203)

```

kai-morich, 2024-02-24:
did the old SerialInputOutputManager terminate and you created a new one after the re-connect?

Chaffey007, 2025-01-07:
I am getting the same "USB get_status request failed" error.

I get this as soon as the SerialInputOutputManager starts >

'onRunError > java.io.IOException: USB get_status request failed'

The permissions are granted and connection to the usb device is successful. I can write data with no problem, but as soon as i try to use the listener, i get this error. Currently on V3.8.1

What am i missing?