mik3y/usb-serial-for-android #399: is it suitable for UNITY

AlexandrVladyka, 2021-11-15:
Hello. Can I use this library for UNITY

kai-morich, 2021-11-16:
please check https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#Does_this_library_work_with_IonicCordovaXamarinApp_Inventor_development_framework

AlexandrVladyka, 2021-11-19:
Can you tell me how to interact with this library in UNITY?

kai-morich, 2021-11-20:
I am not a Unity user. Please follow instructions mentioned in Unity manual, like [this](https://docs.unity3d.com/2021.2/Documentation/Manual/AndroidJavaSourcePlugins.html)

cvasquez-github, 2022-05-24:
Here is an example on how to send a command from Android to Arduino in Unity using this library, hope it helps:
https://github.com/cvasquez-github/unity-android-serial-arduino

mspeirone, 2023-03-02:
Hello. I've implemented this library as a Unity plugin, and all is working well is the Scripting Backend in Unity is set to Mono; but, if I change it to IL2CPP (what is, actually, I need, because I need support for 64bits devices, and upload the app into Google Play), the plugin using this code is not working properly anymore.
The communication speed is too slow, and I lost it just a couple of seconds after establishing connection.

Do you have any suggestions to fix it?

jhclaura, 2023-09-08:
@mspeirone did you find the solution for it? if so would you kindly share your findings?
I'm also interested in using this library as an Unity plugin, with IL2CPP scripting backend. Thanks!

mspeirone, 2023-10-25:
> @mspeirone did you find the solution for it? if so would you kindly share your findings? I'm also interested in using this library as an Unity plugin, with IL2CPP scripting backend. Thanks!

No, I didn't find the solution using this library... we finally implemented it by using our own Android plugin using Android Studio, which finally was integrated into Unity.

designer024, 2024-10-29:
https://github.com/designer024/UnityAndroidSerialPortSample.github.git 

I made the sample, you can try it, hope can help u 
加油!!