mik3y/usb-serial-for-android #404: Sometime reader reads incomplete data

MrRyado, 2021-12-05:
Hi, my code looks something like this, very basic, inspired by the simple USB terminal app repo you did:

```
 private void read() {
        try {
            int len = serialPort.read(buffer, READER_TIMEOUT);
            receive(Arrays.copyOf(buffer, len));
        } catch (IOException e) {
            // when using read with timeout, USB bulkTransfer returns -1 on timeout _and_ errors
            // like connection loss, so there is typically no exception thrown here on error
        }
    }

    private void receive(byte[] data) {
        int receivedDataLength = data.length;
        receivedDataAsString = HexHelper.hex(data);
        System.out.println("Received "+ receivedDataLength +" bytes");
        System.out.println("Data: "+receivedDataAsString);
    }
```

Sometime when reading, I get only first byte, or first X byte. Like 10% of the time. Otherwise I get my complete expected response.

With your app on the playstore USB terminal I always get the complete response.

Any idea how to improve my code? Thank you very much, I am not a very skilled android dev and did my best.

BTW I read the FAQ but I am still wondering.

My baud rate is 11500 (can't change)

kai-morich, 2021-12-05:
please check [this](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time) wiki entry

MrRyado, 2021-12-06:
> please check [this](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time) wiki entry

Thank you, I am reading this wiki entry, however I am wondering how your main app Serial USB terminal always reads correctly. Does it implement automatically a communication protocol? I know my response data uses CRC and after the header there is the number of byte of the data payload. Does your app automatically consider these? 

Thank you very much!

kai-morich, 2021-12-06:
my app does not know about communication protocols. It is permanently reading and fetches the next chunk a few milliseconds later. In your case I recommend to repeat read until complete response (or some larger timeout) is reached.

closed 2021-12-06 by kai-morich

MrRyado, 2021-12-06:
Thank you! I ended up doing what you suggested and it is working.