mik3y/usb-serial-for-android #409: Huge memory leaks freezing device.

Kwazir, 2022-02-08:
Hi, I'm trying to use your library, and wrote this `piece` of code:
```
        val manager = getSystemService(Context.USB_SERVICE) as UsbManager
        val availableDrivers = UsbSerialProber.getDefaultProber().findAllDrivers(manager)
        if (availableDrivers.isNotEmpty()) {
            val driver = availableDrivers[0]
            val connection = manager.openDevice(driver.device)
            if(connection == null) {
                // PERMISSION NOT GRANTED YET
                val usbPermissionIntent = PendingIntent.getBroadcast(this, 0, Intent(INTENT_ACTION_GRANT_USB), 0)
                manager.requestPermission(driver.device, usbPermissionIntent)
                return
            }

            usbSerialPort =
                driver.ports[0].apply {
                    open(connection)
                    setParameters(9600, UsbSerialPort.DATABITS_8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE)
                }
            val usbIoManager = SerialInputOutputManager(usbSerialPort, this)
            usbIoManager.start()
        }
...
        override fun onNewData(data: ByteArray?) = _reading_data_code
        override fun onRunError(e: Exception?) {}

```
I have a device with attached card reader using USB.
When I'm checking what USB device was detected by UsbSerialProber I'm getting the result like below:
vendorId="1027" productId="24577" -> FT232R
what confirms producer data.

Also data parameters used in the command
`setParameters(9600, UsbSerialPort.DATABITS_8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE)`
are taken from the prdevice producer.

I'm using one activity to read data card, so every time destroying it, I'm also executing
`port.close()`

I have to add that I'm reading cards without any problems, so far so good.

Anyway, after 4-6h on this one activity, the device is comepletely freezing.
**Even without 1 card read operation.** Just your library initialized and wait.

I checked the situation using Profiler and the resources used by application is not growing, anyway system is gradually freezing and after some hours the system is freezing.

Can you tell me what should I do, maybe reading periodically from UsbSerialPort to reduce the buffer with data would be a good idea?

my regards,
Michał

kai-morich, 2022-02-08:
FTDI devices send a 2 byte status message each 16msec, but as you have started the usbIoManager, it reads all incomming data, discards the status bytes and does not invoke onNewData().
Maybe it's an issue with your Android device hardware / vendor / version, as the app is not growing. Please try with completely different device.

Kwazir, 2022-02-08:
Shame I can not change my device to something else :(

I'm trying to read the buffer independently to the interface. I'm receiving empty ByteArray every time, but want to check if it change something.

Do you have any other suggestion?