mik3y/usb-serial-for-android #410: Does this lib support hot-plug?

ZhouShowfew, 2022-02-08:
I try use this lib for my device and I manually reset my device to make it disconnected and then connect to android but it failed and I got the error 
`2022-02-08 17:40:35.202 4727-4743/com.queuehop.locationdemo W/SerialInputOutputManager: Run ending due to exception: USB get_status request failed
    java.io.IOException: USB get_status request failed
        at com.hoho.android.usbserial.driver.CommonUsbSerialPort.testConnection(CommonUsbSerialPort.java:152)
        at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:183)
        at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:157)
        at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:224)
        at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:202)
        at java.lang.Thread.run(Thread.java:761)
2022-02-08 17:40:35.202 4727-4743/com.queuehop.locationdemo D/LocationClient: USB get_status request failed
2022-02-08 17:40:35.202 4727-4743/com.queuehop.locationdemo I/SerialInputOutputManager: Stopped
2022-02-08 17:40:35.223 4727-4727/com.queuehop.locationdemo D/UsbConnectionManager: Received intent: Intent { act=android.hardware.usb.action.USB_DEVICE_DETACHED flg=0x10 cmp=com.queuehop.locationdemo/.sensors.UsbConnectionManager (has extras) }
2022-02-08 17:40:35.224 4727-4727/com.queuehop.locationdemo E/UsbConnectionManager: Usb device detached!!!
2022-02-08 17:40:35.225 4727-4727/com.queuehop.locationdemo I/System.out: disconnect
2022-02-08 17:40:35.226 4727-4727/com.queuehop.locationdemo D/UsbDeviceConnectionJNI: close
2022-02-08 17:40:35.226 4727-4727/com.queuehop.locationdemo D/UsbDeviceConnectionJNI: close
2022-02-08 17:40:35.728 4727-4727/com.queuehop.locationdemo D/UsbConnectionManager: Received intent: Intent { act=android.hardware.usb.action.USB_DEVICE_ATTACHED flg=0x10000010 cmp=com.queuehop.locationdemo/.sensors.UsbConnectionManager (has extras) }
2022-02-08 17:40:35.728 4727-4727/com.queuehop.locationdemo D/UsbConnectionManager: Usb device attached!
2022-02-08 17:40:35.730 4727-4727/com.queuehop.locationdemo D/UsbConnectionManager: found location device
2022-02-08 17:40:35.732 4727-4727/com.queuehop.locationdemo I/System.out: connect
2022-02-08 17:40:35.732 4727-4727/com.queuehop.locationdemo I/System.out: 1
2022-02-08 17:40:35.732 4727-4727/com.queuehop.locationdemo D/CdcAcmSerialDriver: trying default interface logic
2022-02-08 17:40:35.732 4727-4727/com.queuehop.locationdemo D/CdcAcmSerialDriver: claiming interfaces, count=2
2022-02-08 17:40:35.732 4727-4727/com.queuehop.locationdemo D/CdcAcmSerialDriver: Control iface=UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=3,mMaxPacketSize=8,mInterval=255]]
2022-02-08 17:40:35.732 4727-4727/com.queuehop.locationdemo D/CdcAcmSerialDriver: data iface=UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=64,mInterval=0]
    UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=64,mInterval=0]]
2022-02-08 17:40:35.732 4727-4727/com.queuehop.locationdemo D/UsbRequestJNI: init
2022-02-08 17:40:35.735 4727-4822/com.queuehop.locationdemo I/SerialInputOutputManager: Running ...
2022-02-08 17:40:44.968 4727-4732/com.queuehop.locationdemo I/art: Do partial code cache collection, code=61KB, data=57KB
2022-02-08 17:40:44.969 4727-4732/com.queuehop.locationdemo I/art: After code cache collection, code=61KB, data=57KB`
and here is my code for connect and disconnect device, by the way I use BroadcastReceiver ACTION_USB_DEVICE_DETACHED and ACTION_USB_DEVICE_ATTACHED for monitor device connect state
`public void connect(UsbSerialDriver driver, UsbDeviceConnection connection) {
        System.out.println("connect");
        this.connection = connection;
        System.out.println(driver.getPorts().size());
        this.port = driver.getPorts().get(0); // Most devices have just one port (port 0)
        try {
            port.open(connection);
            port.setParameters(BAUD_RATE, UsbSerialPort.DATABITS_8,
                               UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);
            port.setRTS(true);
            usbIoManager = new SerialInputOutputManager(port, this);
            usbIoManager.setReadBufferSize(BUFFER_SIZE);
            usbIoManager.start();
        } catch (IOException e) {
            e.printStackTrace();
        }
    }

    public void disconnect() {
        System.out.println("disconnect");
        if (usbIoManager != null) {
            usbIoManager.setListener(null);
            usbIoManager.stop();
            usbIoManager = null;
        }
        if (port != null) {
            try {
                port.setDTR(false);
                port.setRTS(false);
            } catch (Exception e) {
                // e.printStackTrace();
            }
            try {
                port.close();
            } catch (Exception e) {
                // e.printStackTrace();
            }
            port = null;
        }
        if (connection != null) {
            connection.close();
            connection = null;
        }
    }`
    Can you help to see if I have made any mistakes or the lib just does not support this?

kai-morich, 2022-02-08:
If a device is reset, it very likely is re-enumerated by Android like a new device and you have to connect again.

ZhouShowfew, 2022-02-08:
Thanks for the reply! I do connect again after it reset. Here is my code about reconnect
`public void onReceive(Context context, Intent intent) {
        Log.d(TAG, "Received intent: " + intent);
        String action = null;

        if (intent != null) {
            action = intent.getAction();
        }

        UsbManager usbManager = (UsbManager) context.getSystemService(Context.USB_SERVICE);
        UsbDevice device = intent.getParcelableExtra(UsbManager.EXTRA_DEVICE);

        ProbeTable customTable = new ProbeTable();
        customTable.addProduct(0x0483, 0x5740, CdcAcmSerialDriver.class);
        List<UsbSerialDriver> availableDrivers = UsbSerialProber.getDefaultProber().findAllDrivers(usbManager);
        if (availableDrivers.isEmpty()) {
            return;
        }

        for (UsbSerialDriver driver : availableDrivers) {
            if (UsbManager.ACTION_USB_DEVICE_ATTACHED.equals(action)) {
                Log.d(TAG, "Usb device attached!");
                if (identifier.isLocationSensor(device)) {
                    UsbDeviceConnection connection = usbManager.openDevice(driver.getDevice());
                    LocationClient.getInstance().connect(driver, connection);
                }

            }

            if (UsbManager.ACTION_USB_DEVICE_DETACHED.equals(action)) {
                Log.e(TAG, "Usb device detached!!!");
                if (identifier.isLocationSensor(device)) {
                    LocationClient.getInstance().disconnect();
                }
            }
        }`
And as my comment before after connect it just show log
`I/SerialInputOutputManager: Running ...`
but actually no data received, but if I restart my app it will work well.

ZhouShowfew, 2022-02-08:
Thanks for the reply! I do connect again after it reset. Here is my code about reconnect
`public void onReceive(Context context, Intent intent) {
        Log.d(TAG, "Received intent: " + intent);
        String action = null;

        if (intent != null) {
            action = intent.getAction();
        }

        UsbManager usbManager = (UsbManager) context.getSystemService(Context.USB_SERVICE);
        UsbDevice device = intent.getParcelableExtra(UsbManager.EXTRA_DEVICE);

        ProbeTable customTable = new ProbeTable();
        customTable.addProduct(0x0483, 0x5740, CdcAcmSerialDriver.class);
        List<UsbSerialDriver> availableDrivers = UsbSerialProber.getDefaultProber().findAllDrivers(usbManager);
        if (availableDrivers.isEmpty()) {
            return;
        }

        for (UsbSerialDriver driver : availableDrivers) {
            if (UsbManager.ACTION_USB_DEVICE_ATTACHED.equals(action)) {
                Log.d(TAG, "Usb device attached!");
                if (identifier.isLocationSensor(device)) {
                    UsbDeviceConnection connection = usbManager.openDevice(driver.getDevice());
                    LocationClient.getInstance().connect(driver, connection);
                }

            }

            if (UsbManager.ACTION_USB_DEVICE_DETACHED.equals(action)) {
                Log.e(TAG, "Usb device detached!!!");
                if (identifier.isLocationSensor(device)) {
                    LocationClient.getInstance().disconnect();
                }
            }
        }`
And as my comment before after connect it just show log
`I/SerialInputOutputManager: Running ...`
but actually no data was received, but if I restart my app it will work well.

kai-morich, 2022-02-08:
stopping the IoManager is asynchronous. Are you sure everything is closed, IoManager thread is really stopped, before connecting again?

ZhouShowfew, 2022-02-09:
Right, it's asynchronous, I delay the connect 200ms after I get ACTION_USB_DEVICE_ATTACHED Broadcast the connection recovered. Thanks!

ZhouShowfew, 2022-02-09:
`2022-02-09 11:30:04.184 3726-3726/com.queuehop.locationdemo D/UsbConnectionManager: Received intent: Intent { act=android.hardware.usb.action.USB_DEVICE_DETACHED flg=0x10 cmp=com.queuehop.locationdemo/.sensors.UsbConnectionManager (has extras) }
2022-02-09 11:30:04.185 3726-3726/com.queuehop.locationdemo E/UsbConnectionManager: Usb device detached!!!
2022-02-09 11:30:04.186 3726-3726/com.queuehop.locationdemo I/System.out: disconnect
2022-02-09 11:30:04.186 3726-3726/com.queuehop.locationdemo D/UsbDeviceConnectionJNI: close
2022-02-09 11:30:04.186 3726-3726/com.queuehop.locationdemo D/UsbDeviceConnectionJNI: close
2022-02-09 11:30:04.688 3726-3726/com.queuehop.locationdemo D/UsbConnectionManager: Received intent: Intent { act=android.hardware.usb.action.USB_DEVICE_ATTACHED flg=0x10000010 cmp=com.queuehop.locationdemo/.sensors.UsbConnectionManager (has extras) }
2022-02-09 11:30:04.689 3726-3726/com.queuehop.locationdemo D/UsbConnectionManager: Usb device attached!
2022-02-09 11:30:04.702 3726-3726/com.queuehop.locationdemo D/UsbConnectionManager: found location device`

I confirm again found if I don't add delay 200ms its state is also STOPPED before I receiver device connected Broadcast, so I'm sure  IoManager thread is really stopped before connecting again.

closed 2023-09-14 by kai-morich