mik3y/usb-serial-for-android #417: React Native Adapter?

dougbradbury, 2022-02-20:
Has anyone built a react native adapter for this library?

I have been using https://github.com/melihyarikkaya/react-native-serialport, which is backed by https://github.com/felHR85/UsbSerial.

felHR85 has fallen out of maintenance and I am seeing some bugs with prolonged use.

This driver seems like it's got a good contributor group and fairly active work.  I am thinking about building a RN adapter but just wanted to make sure I wouldn't be duplicating someone else's effort.

kai-morich, 2022-02-21:
I am not aware of a wrapper. If you implement it, please add to [usb-serial-for-android wiki](https://github.com/mik3y/usb-serial-for-android/wiki/Projects-Using-usb-serial-for-android).

arwys, 2022-07-29:
@dougbradbury hey I am working a project with react native serial port it seems the module can not handle a large string, I want to send a file to a firmware, it seems some line of codes are missing after I check the file, have you found a good package for react native , especially flushing data from android to a firmware ?

dougbradbury, 2022-07-29:
> @dougbradbury hey I am working a project with react native serial port it seems the module can not handle a large string, I want to send a file to a firmware, it seems some line of codes are missing [...]

@arwys - I have not found one.  Would you have any interest in working on one together?

kai-morich, 2022-07-29:
[bulkTransfer](https://developer.android.com/reference/android/hardware/usb/UsbDeviceConnection#bulkTransfer(android.hardware.usb.UsbEndpoint,%20byte[],%20int,%20int,%20int)) has a size limit on older Android versions, so you have to send in smaller chunks.
to prevent partial data loss on timeout, should be send in native USB packet size chunks as done [here](https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CommonUsbSerialPort.java#L233)

arwys, 2022-08-01:
@dougbradbury that's a good idea 👌

arwys, 2022-08-01:
@kai-morich thank you Kai , I will try that one first, actually I have no knowledges about serial port protocol, and try to combine your method with my module, don't know will work or not, but I will try it and thank you

arwys, 2022-08-05:
@kai-morich hi Kai , i tested App (“Serial USB Terminal “) from playstore and tried to send raw data to my Firmware , but when i check it in my firmware ESP 8266 , there is no data sent. Do you have an idea what is the problem, i used react native serial for from USB Serial by fehr and some line of codes are missing, i think thats happen because of asynchronous process , and i remembered that you taught me to use a bulkTransfer, it seem this methods are exist to at there ,but may i know what timeout value that i should set 
 if i want to send a file up 120 kB  ? Thank you

joshua-smith4, 2023-12-11:
I am going to start on this today. @dougbradbury I'd love your review if this issue is still relevant for you. Also, if you have already put some work into this let me know and I'll do my best to integrate it. Thanks!

dougbradbury, 2023-12-11:
@joshua-smith4  I have not put any work into.  I've been limping along with react-native-serial port with limited success.  Thanks for picking it up!

VictorBarth, 2024-07-29:
Hi everyone. I'm working in a project that I use ESP32 to send jsons to a android device using USB (for now i'm using firebase). The problem is: I have an app that was not made by me (I'm learning RN just for this project), it was made using PWA and EXPO, but I'll only receive data from ESP32 in android devices and wanna know the best way to do that (if its possible). I've seen that I could create a module or API just to include USB, is this the best way? Is it possible? Anyone knows if there's a lib that works with expo? The main problem is obviously that I cannot access manifest.xml or gradle in EXPO, and I don't know if I eject the expo the project could break