mik3y/usb-serial-for-android #418: Does this lib support FT311D?

dllhxll, 2022-02-21:
Hello,
thank you for the great lib!

I found that some serial cables use the FT311D chip, such as https://www.youtube.com/watch?v=u87X0q2nEYM. In this Case, the Android phone communicates with the serial cable in Accessory mode.

I read the "Compatible Devices" and It supports FTDI FT232R, FT232H, FT2232H, FT4232H, FT230X, FT231X, FT234XD.

Does this lib support FT311D?  Or how can I adapt this FT311D chip by using this lib?

Thank you!

kai-morich, 2022-02-21:
This device type is currently not supported as it uses a completely different communication protocol. 
Would have to be added as a new driver using Android AOA protocol like in these [FTDI sample apps](http://www.ftdichip.com/Android/SampleApps.zip)

dllhxll, 2022-02-21:
Thank you for your answer！