mik3y/usb-serial-for-android #419: Abnormal exit

si5haoa, 2022-02-21:
run UsbSerialExamples

si5haoa, 2022-02-21:
appear inevitably 。
I have used usbSerial. jar with the same device before and did not find this problem

kai-morich, 2022-02-21:
have you tried rebooting your Android device?

si5haoa, 2022-02-22:
The same thing happens after restarting the device
Getting messages is getting slower and slower and then crashes

kai-morich, 2022-02-22:
did the App already recieved a lot of data? 
This example app simple adds everything to the TextView, which gets significantly slower the more data is shown. Maybe you simply ran into timeout or out-of-memory issue.

si5haoa, 2022-02-22:
Thank you, I will remove the relevant part of the UI and test it, just send and receive data

si5haoa, 2022-02-23:
Thank you very much, I will delete the UI and receive the data, no problem

closed 2022-02-23 by si5haoa