mik3y/usb-serial-for-android #423: Can this framework support mediatek usb and vcom drivers?

justsilencia, 2022-03-11:
I have been searching high and low for a way to utilize flash software over OTG for mediatek device using an Android host.

This library looks very promising. Would it be possible to use this library in conjunction with mtkclient to detect a target mediatek device in BROM mode?

I hope it's not inappropriate to ask here, but I would gladly pay for this solution to be developed. I currently run mtkclient from termux in an android environment, but it won't detect the target device in BROM mode because the mediatek USB drivers are not available for android OS.

So, I am looking for a custom implementation. If this is possible please let me know. Thank you!

kai-morich, 2022-03-13:
this looks like a completely different use case

justsilencia, 2022-03-14:
> this looks like a completely different use case

That is unfortunate! Well, if you have the expertise to create a solution for my use case, I would be happy to create a freelance contract and pay your rates.

If not, thanks anyways for your response!

closed 2022-05-24 by kai-morich