mik3y/usb-serial-for-android #426: I get "Error writing 20 bytes at offset 0 of total 20" suddenly at runtime at random times. I am unable to figure out the reason for this. Could you help?

KeaganFerrao, 2022-03-30:


kai-morich, 2022-04-01:
do you get the same issue with the sample app or https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal?

b2256334, 2022-04-19:
Hello! 
I also have the same problem, and i use the sample app which you say ( https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal?) ,also get the same problem.
how can i do?
thanks~

kai-morich, 2022-04-19:
which device are you using?

NinjaRat84, 2022-05-30:
I'm using a Pixel 6 Pro

kai-morich, 2022-05-30:
which USB device type are you using?

NinjaRat84, 2022-05-30:
I'm using USB-C on both ends.

kai-morich, 2022-05-30:
which device type (ftdi, pl2303, cdc, ...) are you using?

NinjaRat84, 2022-05-30:
Ohh, sorry... It's CDC :)

kai-morich, 2022-05-30:
does your CDC device read the data you write? I observed timeout errors, when data cannot be written, very likely due to some buffer being full on recieving side.

NinjaRat84, 2022-05-30:
I doesn't seem lite the write succeeds, only that IOException is thrown.
I've verified that the connection is available with proper port, but it fails during write.

Often this issue is resolved by killing the app and re-connect together with starting the app again. (but not always)

kai-morich, 2022-05-30:
do you have connected FTDI and CDC devices at the same time, or does it work with FTDI without issues, and only when you change to CDC device you observe the issues?

kai-morich, 2022-07-20:
does the issue also happen with FTDI and usb-serial-for-android library?

closed 2023-10-18 by kai-morich