mik3y/usb-serial-for-android #427: ControlTransfer with value 0x8 failed:-1

dllhxll, 2022-04-13:
Hi, I hava a problem, Hope the author can give the answer!

The demo usbSerialExamples show error : "ControlTransfer with value 0x8 failed:-1" when I connect to the serial port (TypC - RJ45)

Excecute code: usbSerialPort.open( connection )
Phone version : Vivo iQOO Z3 Android 11 and Oppo realme Q3 Android 11
The vendorId and productId : 067B, 23c3
OTG : I have opened the OTG mode

kai-morich, 2022-04-17:
Are you using a Prolific PL2303GT? Is it working with Linux?

dllhxll, 2022-04-21:
Yes, the serial line(TypeC - RS232) uses Prolific PL2303GT. And it is working with Linux (Android 11).
I connect to the Switch such like this : Phone(TypeC) <------> serial lineA (TypeC - RS232, using Prolific PL2303GT)  <------> serial lineB (RS232 -  RJ45)  <------> Switch(RJ45)
And the serial lineB (RS232 -  RJ45) can work when I connect PC(RS232) and  Switch(RJ45) by using it.

kai-morich, 2022-04-22:
looks like you use an older version of the library, as the message "ControlTransfer with value ..." has slightly changed in recent versions. Please upgrade as PL2303GT is supported since v3.4.0

dllhxll, 2022-05-09:
The version that I use is v3.4.3.

kai-morich, 2022-05-17:
I have a working PL2303GC, but looks like your PL2303GT is not correctly detected. Distinguishing classic PL2303 and PL2303G* looks tricky, as also the Linux kernel has ongoing work there. 

Please post the `rawDescriptor` value used here: https://github.com/mik3y/usb-serial-for-android/blob/cf9bada887479a11a2de3e62102c81f3904b4e97/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/ProlificSerialDriver.java#L293

kai-morich, 2022-05-18:
please post log output created with latest version https://github.com/mik3y/usb-serial-for-android/commit/f997a8b68ab5a53ad4cd34ec1ef44ab609dcc020

dllhxll, 2022-07-06:
I'm so sorry I only replied now.
I used other serial line to avoid this problem.

I will post the log output created with rawDescriptor after I borrow the PL2303GT serial line

Thank you!

kai-morich, 2022-07-22:
duplicate to #439

closed 2022-07-22 by kai-morich