mik3y/usb-serial-for-android #428: Multi-port device: first port opening issue

alexsaenko27, 2022-04-15:
Hi, I have an Android device with 2 serial ports on which I am developing an application using the library. The two ports are associated with the same UsbSerialDriver.

Opening the first port (... driver.getPorts().get(0)) and verifying through the method offered by the library (isOpen()) the opening status of the two ports, it appears that the first is open and the second closed. From here I assume that the second port is still available for use by other applications, but this does not appear to be the case as the second port results as busy. The latter becomes available only when the first port gets disconnected.

Viceversa by opening the second port (... driver.getPorts().get(1)) and repeating the checks carried out in the first case, it appears that the first port remains available for use by other applications, that is, it is not busy.

It has also been noted that, if the second port is used by another application and you try to open the first port through the library, the traffic on the second serial continues to transit until my application is put in the background. Afterwards the second serial port seems to be busy and the other application that had previously opened it cannot use it until, as described above, the first port is explicitly disconnected from my application.

kai-morich, 2022-04-17:
what kind of multiport serial device are you using?

alexsaenko27, 2022-04-17:
Zebra VC8300, it's a vehicle-mounted computer https://www.zebra.com/us/en/products/mobile-computers/vehicle-mounted/vc8300.html

kai-morich, 2022-04-17:
which driver are you using in this library? CDC?

alexsaenko27, 2022-04-19:
It should be FTDI

kai-morich, 2022-04-19:
How do you check that the other port is busy? The library has no related function. 

From screenshot below, you can see that both ports of a FT2232 can be opened in different apps. When opened in the other app, the connection is revoked from the previous app

[image]

kai-morich, 2022-04-27:
How does the External Application check that a port is available? usb-serial-for-android does not offer a function to check if it is already opened somewhere else, it's `isOpen()` only checks if opened by itself. Do you know which library the External Application is using?
With Android USB API one can open an USB device in multiple Apps and _steal_ the connection from the previous App. I suspect one of your involved Apps is doing this by chance and stops at a certain port, so it works in one direction

closed 2022-07-20 by kai-morich