mik3y/usb-serial-for-android #431: v3.4.4 doesn't work with FTDI-FT232H?

david41896, 2022-05-04:
Trying out the "usbSerialExamples" in the v3.4.4 with a FTDI-FT232H device (PID:  0x6014, VID: 0x403).  The sample application can indeed connect to the device and appears ok in writing as well.  However, read operation always returns 0 bytes, for read option of event or direct.  As far as I can tell, the device has I2C interface.  Please comment if such device is supported by v3.4.4 or not.  If possible, how to update the sample code to allow successful read?

david41896, 2022-05-09:
Thanks for the timely response. Your answer helps a lot, truly appreciated! In any event, just wonder if there is a plan to add I2C support in the future?

kai-morich, 2022-05-17:
I2C is not planned

closed 2022-05-17 by kai-morich