mik3y/usb-serial-for-android #433: prolific vid_067B pid_23c3

WanManito, 2022-05-17:
Is this chip supported in the latest version? Looking forward to your reply，I saw this question raised earlier

WanManito, 2022-05-17:
The new version is tested and still doesn't seem to support it

kai-morich, 2022-05-17:
I have a working PL2303GC, but looks like your PL2303GT is not correctly detected. Distinguishing classic PL2303 and PL2303G* looks tricky, as also the Linux kernel has ongoing work there. 

Please post the `rawDescriptor` value used here: https://github.com/mik3y/usb-serial-for-android/blob/cf9bada887479a11a2de3e62102c81f3904b4e97/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/ProlificSerialDriver.java#L293

WanManito, 2022-05-17:
[image]
------
Do you want me to print the value "rawDescriptors"? I can use it normally according to this man's operation, but I don't know if it affects other operations

kai-morich, 2022-05-18:
please post log output created with latest version f997a8b68ab5a53ad4cd34ec1ef44ab609dcc020

kai-morich, 2022-07-22:
duplicate to #439

closed 2022-07-22 by kai-morich