mik3y/usb-serial-for-android #436: Multiple usb-serial enumeration / identification

prsltd, 2022-06-30:
If I have multiple USB-Serial devices how do I identify them correctly. There appears to be no way to find the physical USB port that the device is plugged into, and each time a device is disconnected / reconnected it is assigned a different device name, for example "/dev/bus/usb/001/nnn", where the "nnn" seems to increase each time it is plugged in.

Some of the devices I've tried (FTDI chipset) have a unique serial number which I can use to find the device even if its plugged out and then plugged into a different physical USB port.

Other devices (eg Prolific VID 1659/VID 8963) don't have a unique serial number. How do I keep track of these devices ?

Unfortunately I have no real control over what someone connects to device, so need some solution to this issue.

I'm hoping someone smarter than me has a solution as it's driving me mad !!

Any help appreciated.

kai-morich, 2022-07-01:
Devices without serial number cannot be distinguished, as mentioned in the [wiki](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#can-i-open-multiple-connections-in-one-application)

prsltd, 2022-07-01:
Thanks, I thought that might be the case, but somehow missed the item in the wiki.

closed 2022-07-05 by kai-morich