mik3y/usb-serial-for-android #437: couldn't resolve mik3y/usb-serial-for-android. Android Studio went looking in the jcenter but failed to find it

meganmeiyang, 2022-07-17:
## Root gradle:
[image]

 ## App dependencies:
[image]

## Got error message:
```
Execution failed for task ':app:mergeDebugNativeLibs'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find com.github.mik3y:usb-serial-for-android:v3.4.5.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/com/github/mik3y/usb-serial-for-android/v3.4.5/usb-serial-for-android-v3.4.5.pom
       - https://repo.maven.apache.org/maven2/com/github/mik3y/usb-serial-for-android/v3.4.5/usb-serial-for-android-v3.4.5.pom
       - https://jcenter.bintray.com/com/github/mik3y/usb-serial-for-android/v3.4.5/usb-serial-for-android-v3.4.5.pom
     Required by:
         project :app
```

**I looked it up in jitpack, the configuration is absolutely correct.**

[image]

With or without "v" (in your readme), result is the same - still failed.

## Tried:
- restarted the android studio
- clean project
- rebuild

### None of it worked. 😿

kai-morich, 2022-07-17:
The jitpack repo has to be added in `allprojects` not `buildscripts`

meganmeiyang, 2022-07-17:
got it. solved...

closed 2022-07-17 by kai-morich