mik3y/usb-serial-for-android #438: how can i change stopbit and PARITY

zxy243153431, 2022-07-20:
my project neet to change these two attribute when indata communication.
For example :

kai-morich, 2022-07-20:
parameters can be changed with setParameters when the connection is open, e.g. tested [here](https://github.com/mik3y/usb-serial-for-android/blob/f8e76c9b3b58e0ae4005215d1bb2d9c4cea401d6/usbSerialForAndroid/src/androidTest/java/com/hoho/android/usbserial/DeviceTest.java#L609). 
maybe you have to wait until old data is fully read/written

closed 2022-10-06 by kai-morich

zxy243153431, 2022-10-08:
> parameters can be changed with setParameters repeatedly when the connection is open, e.g. tested [here](https://github.com/mik3y/usb-serial-for-android/blob/f8e76c9b3b58e0ae4005215d1bb2d9c4cea401d6/usbSerialForAndroid/src/androidTest/java/com/hoho/android/usbserial/DeviceTest.java#L609). maybe you have to wait until old data is fully read/written

i tried  and it runs success! thank you very much.