mik3y/usb-serial-for-android #439: PL2303GT (067b:23c3) fails with "ControlTransfer DEVICE_TYPE_T 0x8 failed: -1"

majbthrd, 2022-07-20:
[This post](https://github.com/mik3y/usb-serial-for-android/issues/427#issuecomment-1106045129) implies that "PL2303GT support was added with v3.4.0", but even with the latest at the moment (v3.4.5), usbSerialExamples fails as shown below (screenshot, logcat, and lsusb output).

The device that I'm using is this: https://www.amazon.com/Communication-Transfer-Between-Prolific-Supports/dp/B08FM7C2TS/

Is there any additional diagnostic information that I can provide?  Thanks.

[image]

```
2022-07-20 16:50:06.593 21949-21949/com.hoho.android.usbserial.examples I/serial.example: Late-enabling -Xcheck:jni
2022-07-20 16:50:06.627 21949-21949/com.hoho.android.usbserial.examples E/USNET: USNET: appName: com.hoho.android.usbserial.examples
2022-07-20 16:50:06.630 21949-21949/com.hoho.android.usbserial.examples D/ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
2022-07-20 16:50:06.633 21949-21949/com.hoho.android.usbserial.examples D/ActivityThread: setConscryptValidator
2022-07-20 16:50:06.634 21949-21949/com.hoho.android.usbserial.examples D/ActivityThread: setConscryptValidator - put
2022-07-20 16:50:06.679 21949-21949/com.hoho.android.usbserial.examples V/studio.deploy: Startup agent attached to VM
2022-07-20 16:50:06.679 21949-21949/com.hoho.android.usbserial.examples V/studio.deploy: No existing instrumentation found. Loading instrumentation from instruments-962d6e1a.jar
2022-07-20 16:50:06.687 21949-21949/com.hoho.android.usbserial.examples W/serial.example: DexFile /data/data/com.hoho.android.usbserial.examples/code_cache/.studio/instruments-962d6e1a.jar is in boot class path but is not in a known location
2022-07-20 16:50:06.689 21949-21949/com.hoho.android.usbserial.examples V/studio.deploy: Applying transforms with cached classes
2022-07-20 16:50:06.703 21949-21949/com.hoho.android.usbserial.examples W/serial.example: Redefining intrinsic method java.lang.Thread java.lang.Thread.currentThread(). This may cause the unexpected use of the original definition of java.lang.Thread java.lang.Thread.currentThread()in methods that have already been compiled.
2022-07-20 16:50:06.703 21949-21949/com.hoho.android.usbserial.examples W/serial.example: Redefining intrinsic method boolean java.lang.Thread.interrupted(). This may cause the unexpected use of the original definition of boolean java.lang.Thread.interrupted()in methods that have already been compiled.
2022-07-20 16:50:06.706 21949-21949/com.hoho.android.usbserial.examples D/ActivityThread: handleBindApplication()++ app=com.hoho.android.usbserial.examples
2022-07-20 16:50:06.707 21949-21949/com.hoho.android.usbserial.examples D/CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10268; state: ENABLED
2022-07-20 16:50:06.750 21949-21949/com.hoho.android.usbserial.examples V/GraphicsEnvironment: ANGLE Developer option for 'com.hoho.android.usbserial.examples' set to: 'default'
2022-07-20 16:50:06.750 21949-21949/com.hoho.android.usbserial.examples V/GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
2022-07-20 16:50:06.757 21949-21949/com.hoho.android.usbserial.examples D/LoadedApk: LoadedApk::makeApplication() appContext.mOpPackageName=com.hoho.android.usbserial.examples appContext.mBasePackageName=com.hoho.android.usbserial.examples
2022-07-20 16:50:06.757 21949-21949/com.hoho.android.usbserial.examples D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2022-07-20 16:50:06.781 21949-21949/com.hoho.android.usbserial.examples D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2022-07-20 16:50:06.789 21949-21949/com.hoho.android.usbserial.examples D/ActivityThread: handleBindApplication() --
2022-07-20 16:50:06.797 21949-21974/com.hoho.android.usbserial.examples D/OpenGLRenderer: RenderThread::requireGlContext()
2022-07-20 16:50:06.798 21949-21974/com.hoho.android.usbserial.examples I/AdrenoGLES-0: QUALCOMM build                   : 5eaa426211, I07ee46fc66
    Build Date                       : 10/08/21
    OpenGL ES Shader Compiler Version: EV031.32.02.16
    Local Branch                     : 
    Remote Branch                    : 
    Remote Branch                    : 
    Reconstruct Branch               : 
2022-07-20 16:50:06.798 21949-21974/com.hoho.android.usbserial.examples I/AdrenoGLES-0: Build Config                     : S P 10.0.7 AArch64
2022-07-20 16:50:06.798 21949-21974/com.hoho.android.usbserial.examples I/AdrenoGLES-0: Driver Path                      : /vendor/lib64/egl/libGLESv2_adreno.so
2022-07-20 16:50:06.803 21949-21974/com.hoho.android.usbserial.examples I/AdrenoGLES-0: PFP: 0x016ee197, ME: 0x00000000
2022-07-20 16:50:06.819 21949-21974/com.hoho.android.usbserial.examples D/OpenGLRenderer: RenderThread::setGrContext()
2022-07-20 16:50:06.844 21949-21949/com.hoho.android.usbserial.examples I/DecorView: [INFO] isPopOver=false, config=true
2022-07-20 16:50:06.845 21949-21949/com.hoho.android.usbserial.examples I/DecorView: updateCaptionType >> DecorView@167e9ed[], isFloating=false, isApplication=true, hasWindowControllerCallback=true, hasWindowDecorCaption=false
2022-07-20 16:50:06.845 21949-21949/com.hoho.android.usbserial.examples D/DecorView: setCaptionType = 0, this = DecorView@167e9ed[]
2022-07-20 16:50:06.851 21949-21949/com.hoho.android.usbserial.examples I/DecorView: getCurrentDensityDpi: from real metrics. densityDpi=240 msg=resources_loaded
2022-07-20 16:50:06.859 21949-21949/com.hoho.android.usbserial.examples W/serial.example: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
2022-07-20 16:50:06.860 21949-21949/com.hoho.android.usbserial.examples W/serial.example: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (unsupported, reflection, allowed)
2022-07-20 16:50:06.996 21949-21949/com.hoho.android.usbserial.examples D/CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10268; state: ENABLED
2022-07-20 16:50:07.038 21949-21949/com.hoho.android.usbserial.examples I/ViewRootImpl@b84884f[MainActivity]: setView = com.android.internal.policy.DecorView@167e9ed TM=true
2022-07-20 16:50:07.082 21949-21949/com.hoho.android.usbserial.examples I/ViewRootImpl@b84884f[MainActivity]: Relayout returned: old=(0,0,1200,2000) new=(0,0,1200,2000) req=(1200,2000)0 dur=11 res=0x7 s={true -5476376635848183808} ch=true fn=-1
2022-07-20 16:50:07.085 21949-21974/com.hoho.android.usbserial.examples D/hw-ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
2022-07-20 16:50:07.085 21949-21974/com.hoho.android.usbserial.examples D/OpenGLRenderer: eglCreateWindowSurface
2022-07-20 16:50:07.104 21949-21949/com.hoho.android.usbserial.examples D/AbsListView:  in onLayout changed 
2022-07-20 16:50:07.105 21949-21949/com.hoho.android.usbserial.examples I/ViewRootImpl@b84884f[MainActivity]: [DP] dp(1) 1 android.view.ViewRootImpl.reportNextDraw:11410 android.view.ViewRootImpl.performTraversals:4185 android.view.ViewRootImpl.doTraversal:2911 
2022-07-20 16:50:07.106 21949-21949/com.hoho.android.usbserial.examples D/ViewRootImpl@b84884f[MainActivity]: Creating frameDrawingCallback nextDrawUseBlastSync=false reportNextDraw=true hasBlurUpdates=false
2022-07-20 16:50:07.106 21949-21949/com.hoho.android.usbserial.examples D/ViewRootImpl@b84884f[MainActivity]: Creating frameCompleteCallback
2022-07-20 16:50:07.118 21949-21980/com.hoho.android.usbserial.examples D/ViewRootImpl@b84884f[MainActivity]: Received frameDrawingCallback frameNum=1. Creating transactionCompleteCallback=false
2022-07-20 16:50:07.128 21949-21974/com.hoho.android.usbserial.examples D/ViewRootImpl@b84884f[MainActivity]: Received frameCompleteCallback  lastAcquiredFrameNum=1 lastAttemptedDrawFrameNum=1
2022-07-20 16:50:07.130 21949-21949/com.hoho.android.usbserial.examples I/ViewRootImpl@b84884f[MainActivity]: [DP] pdf(0) 1 android.view.ViewRootImpl.lambda$addFrameCompleteCallbackIfNeeded$3$ViewRootImpl:4987 android.view.ViewRootImpl$$ExternalSyntheticLambda16.run:6 android.os.Handler.handleCallback:938 
2022-07-20 16:50:07.130 21949-21949/com.hoho.android.usbserial.examples I/ViewRootImpl@b84884f[MainActivity]: [DP] rdf()
2022-07-20 16:50:07.130 21949-21949/com.hoho.android.usbserial.examples D/ViewRootImpl@b84884f[MainActivity]: reportDrawFinished (fn: -1) 
[... cut ...]
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

majbthrd, 2022-07-21:
Just in case it might be of additional diagnostic utility, I did a capture of the USB traffic with a USB analyzer.  Per the screenshot below, the PL2303GT is stalling on the wValue = 8 Control Transfer (which is attempted twice in vain by the library):

[image]

kai-morich, 2022-07-21:
please try latest version

majbthrd, 2022-07-21:
I can confirm that it works locally.  Thanks!

closed 2022-07-21 by majbthrd

kai-morich, 2022-07-22:
released with v3.4.6