mik3y/usb-serial-for-android #441: SerialInputOutputManager potential deadlock with no read timout set

rocketraman, 2022-07-23:
The `SerialInputOutputManager` currently potentially does an no-timeout read (depending on the timeout settings) before write in each step. If the read data depends on something being written, this results in a deadlock.

kai-morich, 2022-07-23:
this is already documented as limitation in the code: https://github.com/mik3y/usb-serial-for-android/blob/5bb8db02d5a6f2db32522ce602a6dd7d791de083/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/util/SerialInputOutputManager.java#L147
I recommend to do writing directly in your worker thread

rocketraman, 2022-07-25:
> I recommend to do writing directly in your worker thread

That's totally understood -- in fact I ended up dumping the `SerialInputOutputManager` and doing both writes *and* reads myself to keep things consistent and clean, and I prefer doing the reads and writes in separate threads. Now I am just using usb-serial-for-android to do the serial-specific stuff like line initialization, for which it has been invaluable!

However, the point of this issue was for future users of this library to not have to take this step by changing `SerialInputOutputManager` to avoid the potential deadlock in -- why not use separate reader and writer threads in `SerialInputOutputManager`?

kai-morich, 2025-09-04:
With #615 read and write are separated in own threads