mik3y/usb-serial-for-android #443: How to detect any possible device

dlandtaa, 2022-07-26:
I have an app that needs to detect and list out any possible usb device that is connected. I have tried using `UsbSerialProber.getDefaultProber()` but there are already several test devices I have that it can't seem to detect. I saw [this part o the docs](UsbSerialProber.getDefaultProber()) but I don't want to use a hard coded a list of vendor and product IDs because I don't know all the values (it could be any vendor and product).

Is this possible?

kai-morich, 2022-07-26:
you can get the device list with

        UsbManager usbManager = (UsbManager) getActivity().getSystemService(Context.USB_SERVICE);
        for(UsbDevice v : usbManager.getDeviceList().values())

some devices like mass storage, keyboard, ... are typically filtered out by Android.

dlandtaa, 2022-07-26:
Works well, thank you!

closed 2022-07-26 by dlandtaa

meshileya, 2022-12-22:
hi @kai-morich , thanks so much for this great work as well as support.

I will like to ask if it is possible to also use a USB type B cable for a read and write operation to  a windows application using a specified COM

Jai-GAY, 2022-12-23:
I thought this project is about the phone is the USB host, in your case who is the USB host?
Do you think you have gone to the wrong place to check for an answer?

meshileya, 2022-12-23:
yes, the phone is the USB host.

only trying to establish communication with a windows application.

Jai-GAY, 2022-12-23:
Eh, are you saying you are trying to ask MS Windows OS to be the USB slave?