mik3y/usb-serial-for-android #444: can support usb composite devices？

jiaowoxiaolu, 2022-07-27:
Hi kai-morich
  Thank you for your contribution to this program，and I wonder if this program can support usb composite devices？
I'm using a linux usb composite device, which is composed of RNDIS, ECM and ACM.When I plugged this device into my phone, the phone recognized PID and VID, but there was no driver.I want to know how to write a program to recognize the ACM serial port in this usb composite device and use it normally.

I look forward to your reply.

kai-morich, 2022-07-27:
this library supports composite devices. it checks for appropriate interfaces [here](https://github.com/mik3y/usb-serial-for-android/blob/5bb8db02d5a6f2db32522ce602a6dd7d791de083/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CdcAcmSerialDriver.java#L140)

jiaowoxiaolu, 2022-07-27:
Thank you for your reply.I will give it a try.

closed 2022-08-15 by kai-morich