mik3y/usb-serial-for-android #449: Receive message from multiple ports with correlated port and devices information

sintyasin, 2022-09-13:
Hi mik3y, first of all, thank you for your amazing library.

Let's assume i have a device that can store up to 4devices id and 2ports (so there is 8 ports in total)
Similar to TerminalFragment from your example, i tried to create array of objects to connect to all of my ports and i'm able to write to the correct port as i desire. The problem is when i use receive message, i don't know which message is for which port/device. It's okay if the port/device id is changing over time, i will synchronize it each time a device is reattach.

I tried to create threads for each objects like someone suggested earlier [here](https://github.com/mik3y/usb-serial-for-android/issues/164) . But i always get the received message from main thread.

Can you help me with this? Thank you in advance.

kai-morich, 2022-09-13:
you should create a SerialInputOutputManager per device/port, but instead of making your Fragment implementing SerialInputOutputManager.Listener recieving all data you cannot distinguish any more, you should create additional Listener classes, where you can store device/port as metadata.

sintyasin, 2022-09-14:
It works! thanks a lot, you're a legend 👍

closed 2022-09-14 by sintyasin