mik3y/usb-serial-for-android #451: onNewData Callback Incomplete data

wjgso, 2022-09-17:
Hello！As the title，When I use this “onNewData” to receive data, I often receive incomplete data, and the missing data will enter “onNewData” again.

for example：
I will always receive data, but the data will be like this.
data = "10089"  
first receive ---> "10089"
second receive ---> "1008"
third receive ---> "910089"
Sometimes it is lost, sometimes it is normal

My current processing method is splicing. I would like to ask whether this is a problem with the USB or the library. Looking forward to your reply

wjgso, 2022-09-17:
My version is com.github.mik3y:usb-serial-for-android:3.3.0

kai-morich, 2022-09-17:
This is normal, see https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time

guilhermejsrsantos, 2022-09-19:
@kai-morich Do you have any examples or some guidelines on how to buffer the data being read until a size/count threshold is reached?

guilhermejsrsantos, 2022-09-19:
@kai-morich Do you have any examples on how to implement any of the approaches you've described with event driven data (onNewData)? My application is supposed to receive data at random points in time from a microcontroller and handle it accordingly but I just can't figure out how to implement this behavior with your library. Thanks.

wjgso, 2022-09-22:
> 这是正常的，请参阅 https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time

Thank you for your reply.

wjgso, 2022-09-22:
> @kai-morich Do you have any examples on how to implement any of the approaches you've described with event driven data (onNewData)? My application is supposed to receive data at random points in [...]

closed 2022-09-22 by wjgso