mik3y/usb-serial-for-android #452: "Queueing USB request failed" in Realme C11 running android 11.

yureka, 2022-09-19:
When try to queue the reading request, UsbRequest if throwing following error

W/SerialInputOutputManager: Run ending due to exception: Queueing USB request failed
    java.io.IOException: Queueing USB request failed
        at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:200)
        at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:169)
        at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:225)
        at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:203)
        at java.lang.Thread.run(Thread.java:923)

caoluming, 2022-10-28:
I have the same problem, how did you solve it? thanks

yureka, 2022-10-28:
When the USB Device has hub, UsbRequest is throwing the above error. Without hub, able to read properly.

caoluming, 2022-10-29:
Thank you for your suggestion, I also show that it is related to the hardware here.

caoluming, 2022-10-29:
This is the case with some devices I've detected here, and the frequency of that device's appearance is uncertain. That device, which had never appeared before, appeared suddenly, and the frequency gradually increased. After restarting the MCU, the situation improved, but the device still had this situation, it felt like the cause of the MCU.

kai-morich, 2022-10-29:
I mentioned some reasons here: https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#My_device_sporadically_disconnects_with_codeQueueing_USB_request_failedcode

caoluming, 2022-10-29:
Yes, it is indeed a hardware problem on my side

closed 2023-03-11 by kai-morich