mik3y/usb-serial-for-android #453: no driver found for seeeduino xiao

Renju-P-B, 2022-09-21:
I have burnt simple serial application code into seeeduino xiao. It working well with Arduino serial monitor and linux serial monitor. But when I try to interface with android smartphone using this library, I'm getting no driver error. There is no issue with serial application code, as it works well with arduino uno and this android library. It could be issues related relatively new board of seeduino. I have added pid and vid in the xml files. I have tried some other things but still gives "no driver error"

kai-morich, 2022-09-21:
the xml is only for autostart of App. You should add a custom prober as mentioned [here](https://github.com/mik3y/usb-serial-for-android#probing-for-unrecognized-devices)

Renju-P-B, 2022-09-23:
Thank you. Figured it out.

closed 2022-09-23 by Renju-P-B