mik3y/usb-serial-for-android #458: How to show all ports correctly?

sintyasin, 2022-11-08:
Hi mik3y,

I tried to connect USB serial with 8ports to my computer and it shows 8ports.
But, when i connect it to your example apps "Serial USB Terminal", it only shows 3ports.
How is that possible? Is there any solution or things i could try to fix this?

kai-morich, 2022-11-08:
which kind of USB device are you using?

sintyasin, 2022-11-08:
modem pool 4 port smartcom chipset wavecom Q24 like in this link below

https://shopee.co.id/modem-pool-4-port-smartcom-chipset-wavecom-Q24-i.70401353.18433788696?sp_atk=c5fd415e-e2f1-4501-a222-b537bcbbad67&xptdk=c5fd415e-e2f1-4501-a222-b537bcbbad67

kai-morich, 2022-11-09:
if it does not come with a dedicated driver, it is uses a standard driver. 
if the standard driver is for one of the mentioned chips, it should also work with this library.

kai-morich, 2022-11-09:
please post export from 'USB Device Info' app. The one from Alexandros Schillings. The other app with same name does not collect detaial

sintyasin, 2022-11-14:
Hello, i tried to connect 2 devices to 'USB Device Info'.
The first one which is working well with you library shows some info.
But, the second one (which i have problems earlier) do not show anything, even after i update the DB and click refresh multiple times.

What does this mean? Is it possible that some devices just not compatible and can't be read by Android Serial Driver?

closed 2023-03-11 by kai-morich