mik3y/usb-serial-for-android #461: Drivers is empty with PCF8523 IC

kingfisherphuoc, 2022-12-01:
Hi, I am having an issue with finding the custom driver for my usb-serial device.  This is my `Adafruit` device. As far as I researched, the IC is `PCF8523`. I could not find it on the supported list of custom prober.  My question is: how can I communicate with the device using custom prober?

```
{
/dev/bus/usb/001/002=UsbDevice[mName=/dev/bus/usb/001/002,
mVendorId=9114,
mProductId=32971,
mClass=239,
mSubclass=2,
mProtocol=1,
mManufacturerName=Adafruit,
mProductName=QT Py M0,
mVersion=1.00,
mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@a29ca09, 
mHasAudioPlayback=false, 
mHasAudioCapture=false, 
mHasMidi=false, 
mHasVideoCapture=false, 
mHasVideoPlayback=false,
mConfigurations=
	[
		UsbConfiguration
			[
				mId=1,
				mName=null,
				mAttributes=160,
				mMaxPower=50,
				mInterfaces=
				[
					UsbInterface
					[
						mId=0,
						mAlternateSetting=0,
						mName=null,
						mClass=2,
						mSubclass=2,
						mProtocol=0,
						mEndpoints=
						[
							UsbEndpoint
							[
								mAddress=129,
								mAttributes=3,
								mMaxPacketSize=16,
								mInterval=16
							]
						]
					]
					UsbInterface
					[
						mId=1,
						mAlternateSetting=0,
						mName=null,
						mClass=10,
						mSubclass=0,
						mProtocol=0,
						mEndpoints=
						[
							UsbEndpoint
							[
								mAddress=2,
								mAttributes=2,
								mMaxPacketSize=64,
								mInterval=0
							]
							UsbEndpoint
							[
								mAddress=131,
								mAttributes=2,
								mMaxPacketSize=64,
								mInterval=0
							]
						]
					]	
				]
			]
	]
}
```

kai-morich, 2022-12-01:
`QT Py M0` is based on `SAMD21` CPU so very likely supported by the CDC-ACM. Please add your vendorId + productId to your custom prober

kai-morich, 2023-03-11:
driver should be auto-detected with v3.5.0

closed 2023-03-11 by kai-morich