mik3y/usb-serial-for-android #463: Doesn't detect Arduino Due

phatchman, 2022-12-30:
Arduino Due is not detected by this library. Details for the Due are below:
VID: 9025
PID: 61
Driver Type: CDC

kai-morich, 2023-01-07:
please use a custom prober as depicted [here](https://github.com/mik3y/usb-serial-for-android#probing-for-unrecognized-devices)

phatchman, 2023-01-11:
I'm asking because I want this supported in usb-serial-telnet-server, which uses your library. So I'm not able to use the workaround you suggest and the author there pointed me here.

It seems like this id just needs to be added to the UsbId.java along with all of the other Arduino flavours. It uses the same CDC USB Serial chipset.

kai-morich, 2023-01-11:
the usb-serial-telnet-server only uses default prober. you should add a custom prober [here](https://github.com/ClusterM/usb-serial-telnet-server/blob/9972246d71a3d0aeb4dfb9423c64befbabff3a83/app/src/main/java/com/clusterrr/usbserialtelnetserver/UsbSerialTelnetService.java#L76), e.g. like done in sample app in this repo [here](https://github.com/mik3y/usb-serial-for-android/blob/5bb8db02d5a6f2db32522ce602a6dd7d791de083/usbSerialExamples/src/main/java/com/hoho/android/usbserial/examples/TerminalFragment.java#L200)

phatchman, 2023-01-11:
As it's not my code. I can't do what you are asking.  Can you not add the ID to the  library so that all applications can support the due without having to make a custom prober?

closed 2023-03-03 by kai-morich