mik3y/usb-serial-for-android #465: no driver

GanZzaz, 2023-01-07:
Please teach me how to solve if "no driver" appears under "actuvutt_main". I have added "pid" and "uid" to "device_filter.xml"

GanZzaz, 2023-01-07:
If you add in "CustomProber" it shows no connection

kai-morich, 2023-01-07:
device_filter is only for App auto-start. You have to add correct VID + PID to custom prober

GanZzaz, 2023-01-07:
Where do I need to add to

kai-morich, 2023-01-07:
e.g. [here](https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialExamples/src/main/java/com/hoho/android/usbserial/examples/CustomProber.java)

GanZzaz, 2023-01-07:
我在这里也添加了，能够进入通信界面，可是发送时，弹窗显示"not connected"

GanZzaz, 2023-01-07:
I also added here, you can go to the communication screen, but when you send it, it pops up saying "not connected."

GanZzaz, 2023-01-07:
I added here, I can go to the communication screen, but I can't send it, it pops up and says "not connected."

GanZzaz, 2023-01-07:
The connected in TerminalFragment.java is false

tombrooks1994, 2023-01-20:
I am having the same issue with a CP210x device. I am trying to get it to connect but having no luck @kai-morich are you able to advise?

closed 2023-03-03 by kai-morich