mik3y/usb-serial-for-android #470: Unable to read from ESP32-S2, but writing to it works

vlaminck, 2023-02-07:
### My Setup
I have an ESP32-S2 board connected to my Android device via an OTG usb hub. Let's call this `BoardA`. This ESP32-S2 board communicates to another ESP32-S2 board that is connected to my computer. Let's call this `BoardB`. I have `BoardB` connected to my computer so I can see the output in the Arduino IDE serial monitor. (_Quick sanity check, when I have both boards connected to the serial monitor, they both show output as expected so I know `Serial.write` and `Serial.println` work on both boards._)

In my Android code, I connect to `BoardA` using this prober
```
val customTable = ProbeTable()
customTable.addProduct(0x1A86, 0x55D4, CdcAcmSerialDriver::class.java) // ESP32 V2
customTable.addProduct(0x239A, 0x80EB, CdcAcmSerialDriver::class.java) // ESP32-S2
val prober = UsbSerialProber(customTable)
```
I have an ESP32 V2 board that works just fine, but I'm trying to migrate to the S2 board which is how I found this issue.

### How My Setup Works
1. The Android device writes a message to `BoardA`
2. `BoardA` writes to Serial to let Android know it received the message
3. `BoardA` passes the message to `BoardB`
4. `BoardB` writes to Serial (observed via Arduino IDE Serial Monitor)
5. `BoardB` responds to `BoardA` that the message was handled
6. `BoardA` writes to Serial to let Android know the loop was successful

### Observations
* When using an ESP32-S2 board, Android does not receive any serial communication. It is expected to receive a message in steps 2 and 6 above. I do see my messages in the Serial Monitor in step 4 so I know I can write to the board just fine.
* When using an ESP32 V2 board, Android receives serial communication in steps 2 and 6 above.
* When both ESP32-S2 boards are connected to the Serial Monitor, I see all the expected output from both boards so I know both of them can write to serial just fine.

### Final Thoughts
I feel like the problem is with the ESP32-S2 board and not usb-serial-for-android, but the fact that I can see output in the Serial Monitor leads me to believe that the board does in fact write to Serial just fine. I'm not really sure what to try next, and would love some help if you have any ideas.

I upload to my board using Arduino IDE 2.0.3 and [arduino-esp32](https://github.com/espressif/arduino-esp32) version 2.0.6.

kai-morich, 2023-02-08:
please check: https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-I_am_using_an_Arduino_Uno_Sparkfun_Pro_Micro_or_other_Arduino_but_codeSerialwritecode_does_not_send_data

vlaminck, 2023-02-08:
Thank you! I figured I was overlooking something simple.

closed 2023-02-08 by vlaminck