mik3y/usb-serial-for-android #473: Cannot receive data

NortacYWang, 2023-02-23:
I am trying to retrieve the data from USB,  the command is when USB received string "SA+SAQ", it will return related data(string) to the android device.

I have coded the write and read like this: 

connect part
```
    @ReactMethod
    public void connect(Integer baudRate, Integer dataBits, Integer stopBits, Integer parity, Promise promise) {
        UsbManager usbManager = getUsbManager();
        UsbSerialDriver firstDriver = getFirstSerialDevice();
        if (firstDriver == null) {
            promise.resolve("no device);
            return;
        }
        UsbDeviceConnection connection = usbManager.openDevice(firstDriver.getDevice());
        if (connection == null) {
             promise.resolve("conect fails");
            return;
        }
        this.usbSerialPort = firstDriver.getPorts().get(0);
        try {
            usbSerialPort.open(connection);
            usbSerialPort.setParameters(baudRate, dataBits, stopBits, parity);
            connected =  true;
            promise.resolve("OK");
        } catch (IOException e) {
            promise.resolve("open fails");
        }
    }
```

I got "OK" which means the usb is connected,

write and read part
```
@ReactMethod
    public void sendAndReceive(String data, Promise promise) {
        if (this.usbSerialPort == null) {
            promise.resolve("no device");
        }
        byte[] dataBytes;
      
        dataBytes = data.getBytes();  // data is SA+SAQ

      try{
       usbSerialPort.write(dataBytes, 2000);
      } catch(IOException e) {
      promise.resolve("write fails")
      }
      

       try{
            byte buffer[] = new byte[16];
            int numBytesRead = usbSerialPort.read(buffer, 20000);
            String texto = new String(buffer);
            promise.resolve(texto);
        } catch (IOException e) {
            e.printStackTrace();
            promise.resolve(e.toString());
        }

```

the texto always returns empty , even the usb is coded when invalid command received, it will return string "invalid".

can anyone give me some suggestions? I have worked for this project for several days.. got no clue

Thank you very much

kai-morich, 2023-02-24:
Please check https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-How_to_size_the_codereadcode_buffer

NortacYWang, 2023-02-24:
Thank you Kai, I have tried to setReaderBufferSize in  SerialInputOutputManager, but still got no data.

I also tried the read code in the example: 
```
    try {
            byte[] buffer = new byte[8192];
            int len = usbSerialPort.read(buffer, READ_WAIT_MILLIS);
            receive(Arrays.copyOf(buffer, len));
        } catch (IOException e) {
            // when using read with timeout, USB bulkTransfer returns -1 on timeout _and_ errors
            // like connection loss, so there is typically no exception thrown here on error
            status("connection lost: " + e.getMessage());
            disconnect();
        }
```
but still no data received. btw I have tried the android usb serial terminal to test, it did return the string "invalid command". and I am using ReactNative with this library

Thank you (maybe someone can suggest me a RN library? )

NortacYWang, 2023-02-28:
I have fixed this by override onNewData function in ```SerialInputOutputManager.Listener ``` instead of using ```usbSerialPort.read``` after write to usb immediately

closed 2023-02-28 by NortacYWang