mik3y/usb-serial-for-android #476: Not Able to read data getting errors no drivers

devendranegi121, 2023-03-06:
We are not able to connect read data from cdc devices sowing no drivers 
[image]
Cdc device 

Device info
[image]

Can you please guide me what wrong I am doing

kai-morich, 2023-03-07:
looks like you didn't use the customer prober. Please try adding to existing custom prober in `app/src/main/java/de/kai_morich/simple_usb_terminal/CustomProber.java`

devendranegi121, 2023-03-07:
> looks like you didn't use the customer prober. Please try adding to existing custom prober in `app/src/main/java/de/kai_morich/simple_usb_terminal/CustomProber.java`

Same file I have updated with device info. could you please help me find issue
class CustomProber {

    static UsbSerialProber getCustomProber() {
        ProbeTable customTable = new ProbeTable();
        customTable.addProduct(0x2F94, 0x0007, CdcAcmSerialDriver.class); // e.g. Digispark CDC
        return new UsbSerialProber(customTable);
    }

}

kai-morich, 2023-03-11:
with v3.5.0 you do not need a custom-prober any more

devendranegi121, 2023-03-31:
@kai-morich , It's working now
Thank you so much for your Quick support