mik3y/usb-serial-for-android #489: opening project 3.5.1

Jai-GAY, 2023-04-13:
Hi

Does anyone successfully open the [latest release project](https://github.com/mik3y/usb-serial-for-android/releases/tag/v3.5.1) in Android Studio Electric Eel?

A problem occurred configuring root project 'usb-serial-for-android-3.5.1_org'.
> Could not resolve all files for configuration ':classpath'.
   > Could not resolve com.android.tools.build:gradle:7.4.2.
     Required by:
         project :
      > No matching variant of com.android.tools.build:gradle:7.4.2 was found. The consumer was configured to find a runtime of a library compatible with Java 8, packaged as a jar, and its dependencies declared externally, as well as attribute 'org.gradle.plugin.api-version' with value '7.5' but:
          - Variant 'apiElements' capability com.android.tools.build:gradle:7.4.2 declares a library, packaged as a jar, and its dependencies declared externally:
              - Incompatible because this component declares an API of a component compatible with Java 11 and the consumer needed a runtime of a component compatible with Java 8
              - Other compatible attribute:
                  - Doesn't say anything about org.gradle.plugin.api-version (required '7.5')
          - Variant 'javadocElements' capability com.android.tools.build:gradle:7.4.2 declares a runtime of a component, and its dependencies declared externally:
              - Incompatible because this component declares documentation and the consumer needed a library
              - Other compatible attributes:
                  - Doesn't say anything about its target Java version (required compatibility with Java 8)
                  - Doesn't say anything about its elements (required them packaged as a jar)
                  - Doesn't say anything about org.gradle.plugin.api-version (required '7.5')
          - Variant 'runtimeElements' capability com.android.tools.build:gradle:7.4.2 declares a runtime of a library, packaged as a jar, and its dependencies declared externally:
              - Incompatible because this component declares a component compatible with Java 11 and the consumer needed a component compatible with Java 8
              - Other compatible attribute:
                  - Doesn't say anything about org.gradle.plugin.api-version (required '7.5')
          - Variant 'sourcesElements' capability com.android.tools.build:gradle:7.4.2 declares a runtime of a component, and its dependencies declared externally:
              - Incompatible because this component declares documentation and the consumer needed a library
              - Other compatible attributes:
                  - Doesn't say anything about its target Java version (required compatibility with Java 8)
                  - Doesn't say anything about its elements (required them packaged as a jar)
                  - Doesn't say anything about org.gradle.plugin.api-version (required '7.5')

kai-morich, 2023-04-13:
Please try changing java version for the build tools as mentioned here: https://stackoverflow.com/questions/66980512/android-studio-error-android-gradle-plugin-requires-java-11-to-run-you-are-cur

Jai-GAY, 2023-04-13:
Thanks for the response and I manage to resolve it by modifying the [gradle-wrapper.properies](https://github.com/mik3y/usb-serial-for-android/blob/9bd1f257739f57a9079c15cd25ea4331fb587091/gradle/wrapper/gradle-wrapper.properties#L6) file to the following.

distributionUrl=https\://services.gradle.org/distributions/gradle-7.5.1-**bin**.zip
Or
distributionUrl=https\://services.gradle.org/distributions/gradle-7.5-**bin**.zip

now only an unimportant warning.

Software Components will not be created automatically for Maven publishing from Android Gradle Plugin 8.0. To opt-in to the future behavior, set the Gradle property android.disableAutomaticComponentCreation=true in the `gradle.properties` file or use the new publishing DSL.

closed 2023-04-13 by Jai-GAY