mik3y/usb-serial-for-android #491: USB get_status request failed

WanManito, 2023-04-23:
version = 3.5.1

When I connect to a USB serial port with no available devices, I can connect normally, but when I close it, it will trigger the exception "java. io. IOException: USB get_status request failed", and I cannot connect all devices normally. If I connect to new available devices again, the exception "Reset failed: result=-1" will always be triggered

kai-morich, 2023-04-23:
which device type are you using? do you get the exception from the close command?

WanManito, 2023-04-23:
Ftdi vid = 1027，pid = 24577
prolific vid = 1659,pid = 9155,
close exception  = "java. io. IOException: USB get_status request failed"

WanManito, 2023-04-23:
As a supplementary explanation, I used two types of USB serial devices with the same phenomenon. The testing process is as follows:
The USB port connects to Android devices, but DB9 is not connected to any devices. It can open the connection normally, and closing it again will trigger an exception, and then it cannot connect to any other devices. @kai-morich

kai-morich, 2023-04-25:
does it fail with FTDI or Prolific device? Prolific devices are known for unstable connection.

WanManito, 2023-04-26:
Hello, both devices have been tested and this phenomenon is inevitable. I'm not sure if I have described it clearly @kai-morich

WanManito, 2023-04-26:
[image]

kai-morich, 2023-04-26:
when opening again, you very likely cannot reopen an existing `UsbSerialPort` object, but have to redo initialization starting with  `manager.openDevice`

WanManito, 2023-04-26:
[image]

WanManito, 2023-04-26:
I'm very sorry, because I used a USB hub to connect two USB serial devices. After repeated testing, it was a problem with the USB hub. I switched to another brand of USB hub and everything was normal. Thank you for your support

closed 2023-04-26 by WanManito

Jai-GAY, 2023-04-26:
@WanManito 

I think is good that you share the brand, model and a picture so the rest of the users can avoid it to repeat the same questions to the developers.