mik3y/usb-serial-for-android #492: no driver for device

Ranjit8298, 2023-04-24:
Hii @all , Please help
How to resolve this issue no driver for device?

narendran12, 2023-05-08:
Hi @kai-morich @mik3y i am using the latest library. When i connect the usb pendrive is detecting and it says no driver, on tap of that i am getting error toast like no driver. Could you please help.

@Ranjit8298 Have u got any workaround for this issue?

[image]

kai-morich, 2023-05-08:
your 'pendrive' is a memory device, but not a serial device, so you cannot open it with this library

Jai-GAY, 2023-05-09:
> connect the usb pendrive is detecting
Do not quite understand why you need this app to open USB mass storage class device where all OTG phones can use the default file explorer to browse the contents.

[this app compatible-devices](https://github.com/mik3y/usb-serial-for-android#compatible-devices)

closed 2023-05-26 by kai-morich