mik3y/usb-serial-for-android #493: Need Support for security fix

madhuri3195, 2023-04-25:
Hi,

I use the dependency for my Android project and build the apk. when I share the apk with the help of Gmail, it is detect it is a virus so I scan that, 3 antiviruses detect it detects a virus. and I was scan particular usb_serial_for_android .zip file through a virus and here is the link
https://www.virustotal.com/gui/file/1cc763c7a18f82c994e17735d1a04e955922fd36c6687a626bebcc0b70be7512/detection

kai-morich, 2023-04-25:
only 1 of 61 tools marked this as virus, so more likely is a false positive.

where do you got the usb_serial_for_android.zip from? I am not aware that this library / jitpack provides zip files.

madhuri3195, 2023-04-26:
The version i have used is  { groupId 'com.github.mik3y'
                artifactId 'usb-serial-for-android'
                version 'v3.5.0beta' } and used this code[(https://github.com/mik3y/usb-serial-for-android/archive/refs/tags/v3.5.0.zip. )](URL) and facing the same issue.

kai-morich, 2023-04-26:
If there would an explanation what the virus report complains about, it could be adressed. 
Please contact McAfee support to get more details. 

Nevertheless I asume it is a false positive as only 1 of 61 tools marks this code.

Jai-GAY, 2023-04-26:
the Window Defender and Symantec 14.3 RU5 scan both downloaded [https://github.com/mik3y/usb-serial-for-android/archive/refs/heads/master.zip](https://github.com/mik3y/usb-serial-for-android/archive/refs/heads/master.zip) and [https://github.com/mik3y/usb-serial-for-android/archive/refs/tags/v3.5.1.zip](https://github.com/mik3y/usb-serial-for-android/archive/refs/tags/v3.5.1.zip), do not detect any issue.

closed 2023-05-26 by kai-morich