mik3y/usb-serial-for-android #495: can not write data to pico

hfutzhu, 2023-05-14:
can not write data to pico. but the 'write' function execute successfully.

kai-morich, 2023-05-26:
please provide more details. which driver type are you using?

Limocello, 2023-06-02:
I was working on the same Issue yesterday and for me the Issue was that I didn't set the DTR flag. My code is very messy and I still wasn't able to implement an input listener but I initialize the Connection to the Pico like this:

`    private void connect2(Boolean permissionGranted){

        UsbDevice device = null;
        UsbManager usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
        for(UsbDevice v : usbManager.getDeviceList().values())

            if(v.getDeviceId() != 0) //This uses the first Device available. Statement should be replaced by the DeviceId (2004 for me)
                device = v;

        UsbSerialDriver driver = UsbSerialProber.getDefaultProber().probeDevice(device);
       
       /* Pico doesn't need Custom Prober
        if(driver == null) {
            System.out.println("Using Custom Prober...");
            driver = getCustomProber().probeDevice(device);
        }
        */
        if(driver == null) {
            status("connection failed: no driver for device");
            return;
        }
        

        usbSerialPort = driver.getPorts().get(0); //TODO: Go through these Ports to see which is right
        UsbDeviceConnection usbConnection = usbManager.openDevice(driver.getDevice());

        if(usbConnection == null){
            int flags = Build.VERSION.SDK_INT >= Build.VERSION_CODES.M ? PendingIntent.FLAG_MUTABLE : 0;
            PendingIntent usbPermissionIntent = PendingIntent.getBroadcast(MainActivity.this, 0, new Intent(BuildConfig.APPLICATION_ID + ".GRANT_USB"), flags); 
            usbManager.requestPermission(driver.getDevice(), usbPermissionIntent);
            usbConnection = usbManager.openDevice(driver.getDevice());
            if(usbConnection == null){
                send("hard fail Were the Permissions not given?");
            }

        }
        connected = Connected.Pending;
        try {
            usbSerialPort.open(usbConnection);
            usbSerialPort.setParameters(19200, UsbSerialPort.DATABITS_8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);

            usbSerialPort.setDTR(true); // for arduino and raspberrypi, ...
            //usbSerialPort.setRTS(true);
            
            //Testing 
            byte[] message = "           ".getBytes();
            int len = 0;
            usbSerialPort.write("Test".getBytes(StandardCharsets.UTF_8), 50);
            len = usbSerialPort.read(message, 50);
            status(new String(message) + "length: " + len);
            status(String.valueOf(usbSerialPort.isOpen()));
        } catch (Exception e) {
            status("connection lost: " + e.getMessage());
            disconnect();
        }

    }`

Please note that I'm in the middle of debugging and currently really struggling with reading data myself. But you should be able to read and write with this setup. the status(str message) function is just a function that I'm using to log the data.

closed 2023-10-18 by kai-morich