mik3y/usb-serial-for-android #496: How can I use this library to read data from serial device in background when the terminalfragment is onPause?

uniqueram, 2023-05-15:
I have made a custom keyboard which is  a service and I need to read obtained data via keyboard using this library I could only get data to the keyboard when the terminalfragment is open and read data on UI but I need data when the terminalfragment is onPause.

kai-morich, 2023-05-15:
you need a background service as used [here](https://github.com/kai-morich/SimpleUsbTerminal)

uniqueram, 2023-06-28:
does this application not working in android version 13+?

kai-morich, 2023-06-28:
did you try the background service? 
did do you observe a specific issue on android 13+?

uniqueram, 2023-06-29:
yes, I did and worked on Android 11 but when I try on Android 13+ when data arrive from the serial device application crashes.

kai-morich, 2023-07-03:
what kind of crash do you get?

uniqueram, 2023-07-04:
I don't think the issue is like #494 I have tried that solution everything is working only when Data arrives from the serial device it gets crashed.

closed 2025-04-06 by kai-morich