mik3y/usb-serial-for-android #499: More than one data interface on CDC-ECM

Sa-f43, 2023-05-25:
We are using a CDC-ECM composite device, and it looks like that the CDC Ethernet Control Model can have more than one data interface associated with it for different settings. For that reason, we would end up writing into the wrong interface. This maybe an issue specific to CDC-ECM devices. 
We think If the openInterface() function in CdcAcmSerialDriver does not increment the interfaces count for repeated interface IDs then that may fix the issue. 

[image]

kai-morich, 2023-05-26:
as USB IAD descriptors are not exposed on Java level it's difficult to to group interfaces. The library already has a workaround for RNDIS interfaces. Here would have to handle the `mAlternateSettings`

kai-morich, 2023-05-26:
please provide complete interface list.
probably could add a list of already seen `mId` and skip the the alternate interfaces

Sa-f43, 2023-05-26:
Here is the complete interface list
[image]

kai-morich, 2023-07-16:
@Sa-f43 could you provide the full USB descriptor, e.g. by adding

    Log.d("terminal","USB descriptor:\n" + HexDump.dumpHexString(usbConnection.getRawDescriptors()));
before

    usbSerialPort.open(usbConnection);
in `usbSerialExamples/src/main/java/com/hoho/android/usbserial/examples/TerminalFragment.java` ?

closed 2023-08-24 by kai-morich