mik3y/usb-serial-for-android #500: Not receiving any data

gautam-paliwal, 2023-05-26:
I am tring to communicate between android app and Xbee3. I am able to send data from android app to XBee3 but not receiving any Data from Xbee3 to android app. I have tried to implement SerialInputOutputManager.Listener and int len = port.read(packet, 1000); but none of them works

kai-morich, 2023-05-26:
please try: https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-I_am_using_an_Arduino_Uno_Sparkfun_Pro_Micro_or_other_Arduino_but_codeSerialwritecode_does_not_send_data

Alfaizkhan, 2023-05-31:
I'm also not getting the data in the application sometime.

I have to open application again, make a connection and then only it will work.

Alfaizkhan, 2023-05-31:
@kai-morich 
Any solution or tips?
Why it's happening.

I get nothing in onNewData() sometimes.

closed 2023-09-02 by kai-morich