mik3y/usb-serial-for-android #506: Receiving bytes in wrong order or replaced bytes

Sir-Ask-A-Lot, 2023-07-03:
Hi,

Sometimes a byte I receive is replaced, by what appears to be a 'double' byte, for example I receive '... 31 00 00 00 78 ...' instead of '... 31 36 00 00 78 ...'. Do you have any idea why this could be the case?

I am using using the CdcAcmSerialDriver for the Device I am connecting. If I connect this device to a Windows PC the Cdc-driver is also used, I detect no problems when using Windows.

kai-morich, 2023-07-04:
if the serial device is powered over USB, it might work less stable when powered by the Android device

Sir-Ask-A-Lot, 2023-07-06:
I now have the device from powered a different source with enough juice, but the problem still remains.

closed 2023-09-14 by kai-morich

Sir-Ask-A-Lot, 2023-12-07:
It looks this is a Linux driver problem.