mik3y/usb-serial-for-android #508: Cannot resolve BuildConfig symbol

amarz2195, 2023-07-04:
Hello guys. I am using the github project to build a serial connection.
I use the gtihub as a module imported into my project. Have been following the issue that is resolved by other users. It's working at first. But when I import it inside the project, the same problem occurs which is 'can't find symbol class BuildConfig'. do help me.
[image]
This is my error
[image]
This is my buildconfig that is generated
How can I do this?

amarz2195, 2023-07-04:
[image]
this is the buildconfig that is generated after building the project

amarz2195, 2023-07-04:
why I cannot import import com.hoho.android.usbserial.BuildConfig;?

amarz2195, 2023-07-04:
I have work out the code until I only got this error
[image]
help me on how to get the buildconfig. this is after i added the dependencies in the build.gradle (usbSerialExample)
the java for buildconfig is not generated automatically after adding the dependencies. why?

kai-morich, 2023-07-04:
if your project is using gradle 8.0 you have to add the following line to `gradle.properties` file

    android.defaults.buildfeatures.buildconfig=true

amarz2195, 2023-07-04:
thank you for the information
what does this mean?
[image]

kai-morich, 2023-07-04:
I typically try build->Clean, Build->Rebuild, File->Sync-Project or File->Invalidate-Caches. Sooner or later the issue is resolved.

amarz2195, 2023-07-04:
i have tried all the methods above, all of the method can't help in resolving the error that i get
[image]
it says that because I have two buildconfig files. but the two is in buildconfig java is in different files. one in usbserialexample and also in usbforandroid. how can i do this?

kai-morich, 2023-07-05:
I recommend you restart with a plain fork and then change step by step

amarz2195, 2023-07-05:
thank you sir for the knowledge shared.
I have successfully built the project.
There's some question maybe out of the topic of usb serial for android
After successfully building the project, I tried to run the project with the built. The error shows no module for the main app
[image]
Please help me sir. I am new to this

amarz2195, 2023-07-05:
thank you sir for the assist
have successfully built the project
one more question, when i try running the project, it appears that there are no module appear for the project built
[image]
what should i do? 
I am new to this software.

amarz2195, 2023-07-05:
thank you sir

closed 2023-07-05 by amarz2195